esp8266/Arduino #1258: core_esp8266_main.cpp issue

agtabesh, 2015-12-20:
Hi there,
When I compile examples I get this error:
```
/Users/xxxxx/Documents/Arduino/libraries/esp8266/core_esp8266_main.cpp: In function 'void esp_schedule()':
/Users/xxxxx/Documents/Arduino/libraries/esp8266/core_esp8266_main.cpp:76:38: error: 'ets_post' was not declared in this scope
     ets_post(LOOP_TASK_PRIORITY, 0, 0);
                                      ^
/Users/xxxx/Documents/Arduino/libraries/esp8266/core_esp8266_main.cpp: In function 'void user_init()':
/Users/xxxxx/Documents/Arduino/libraries/esp8266/core_esp8266_main.cpp:149:24: error: 'ets_task' was not declared in this scope
         LOOP_QUEUE_SIZE);
                        ^
Error compiling.
```
Please help

igrr, 2015-12-20:
That's weird. These two functions are declared in `ets_sys.h`:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/ets_sys.h#L175-L176

`main.cpp` includes `ets_sys.cpp` near the top:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_main.cpp#L27

agtabesh, 2015-12-20:
I remove esp8266 from Board Manager and reinstall it
Now I have ets_sys.h in _sdk/include_ but I does not work!
Please help

Links2004, 2015-12-20:
@agtabesh what core version do you use?
RANDOM_REG32 is in there since 12day, so its only on the git version of the core.
https://github.com/esp8266/Arduino/commit/02e22b38cb17697d43f347478d63530c72b61e7d

@igrr may its time for a new rc version.

agtabesh, 2015-12-20:
Ok It works now
but I have [this](https://github.com/Links2004/arduinoWebSockets/issues/29#issuecomment-166115246) issue now

igrr, 2015-12-22:
@Links2004 true, but i would like to get at least some of these fixed before the release:
[unresolved bugs for 2.1.0](https://github.com/esp8266/Arduino/issues?utf8=%E2%9C%93&q=is%3Aopen+is%3Aissue+milestone%3A2.1.0+label%3Abug+-label%3Astaged-for-release+)
Closing this for now.

closed 2015-12-22 by igrr