esp8266/Arduino #1260: Debug statements in the HTTP-Client cause panics

nhahn, 2015-12-20:
While I was trying to debug some of the issues I was having talking with my server using the HTTP client, I was getting a number of unexplainable panics. However, once I turned the debugging off, I noticed that the panics were gone (turning off the debug statements was the only change I made). I think with certain responses, it's causing some type of memory error when trying to log the information out over the serial port. 

Instance 1: 
```
[HTTP-Client][begin] url: http://192.168.1.222:8080/api/devices/simultaneously-predictable-imbecile/newData
[HTTP-Client][begin] host: 192.168.1.222 port: 8080 url: /api/devices/simultaneously-predictable-imbecile/newData https: 0 httpsFingerprint: 
[HTTP-Client] connect http...
[HTTP-Client] connected to 192.168.1.222:8080
[HTTP-Client][sendRequest] Stream payload written: 112
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: Express'
[HTTP-Client][hand
Exception (2):
epc1=0x3fff4d30 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff4d30 depc=0x00000000

ctx: sys 
sp: 3ffef978 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffefb18:  3fff4d30 3ffec89c 3ffec89c 00000001  
3ffefb28:  3ffee804 003a0018 0000ffec 3ffec250  
3ffefb38:  00000000 3fff08a8 00000100 00000000  
3ffefb48:  00000000 00000000 00000000 00000000  
3ffefb58:  00000000 00000000 00000000 00000000 
(.... the stack trace is very long, I didn't paste the whole thing)  
```

Instance 2:
```
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 20'
[HTTP-Client][handleHeaderResponse] RX: 'ETag: W/"14-hLTFDi7nVYXqZWgxcHgpag"'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Sat, 19 Dec 2015 09:40:31 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Conne
Panic /Users/nhahn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266/core_esp8266_main.cpp:114 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  4021d106 3fffdab0 00000000 3fff1cb0  
3fffff90:  00000000 3fffdc20 3fff1ca0 4020fbfd  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
```

Instance 3: 
```
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 20'
[HTTP-Client][handleHeaderResponse] RX: 'ETag: W/"14-hLTFDi7nVYXqZWgxcHgpag"'
[HTTP-Client][handleHeaderResponse
Panic /Users/nhahn/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266/core_esp8266_main.cpp:85 __yield

ctx: sys 
sp: 3fff1cb0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fff1e60:  3fff3dd0 3fff1f2e 4020fc2c 3fff1cb0  
3fff1e70:  00000000 00000000 3fff1bdc 4020fcb0  
3fff1e80:  00000000 00000065 3fff1bdc 4020f2ca  
3fff1e90:  3fff1bdc 00000022 3fff1f02 40213458  
3fff1ea0:  4021cbfe 00000000 00000000 00000050  
```

igrr, 2015-12-22:
Could you please share the sketch to reproduce this?

igrr, 2016-02-29:
HardwareSerial has been reworked and doesn't use `yield()` now, so this issue should not be present in 2.1.0.

closed 2016-02-29 by igrr