esp8266/Arduino #1261: WiFi.softAP doesn't seem to honor ssid or password

develpr, 2015-12-20:
I'm using the "staging" build.

I'm using the example sketch [provided here in the repo](https://github.com/esp8266/Arduino/blob/342c4ae6fb847bfc787f80b89a2bb888d942dc32/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino) which when run should result in a AP being created with a particular name, and secured with a password.

Instead, the AP is created with a generic SSID ("ESP_XXXXXXX") and w/o a password. 

I'm guessing it doesn't matter, but this is on a ESP-12E.

igrr, 2015-12-20:
See #1254

develpr, 2015-12-20:
Silly question maybe, but I'd guess staging would have the latest fixes? Will this be pushed to the stagingrelease/ tag sometime, or should that branch not be used?

igrr, 2015-12-20:
Yeah, staging will get updated shortly.

develpr, 2015-12-20:
Awesome! Is there any documentation about which features are available in staging vs stable? Somebody recently told me that i should be using staging buy obviously it send to behind in certain bug fixes?

igrr, 2015-12-20:
You can check git tags 2.0.0 and 2.0.0-rc2 and commits between them. Also see docs:
http://esp8266.github.io/Arduino/versions/2.0.0/
http://esp8266.github.io/Arduino/versions/2.0.0-rc2/

develpr, 2015-12-20:
Awesome! I'll RTFM ;) - thanks again, and sorry for the duplicate issue (I looked but somehow missed the duplicate issue!)

closed 2015-12-20 by develpr

xetra11, 2016-09-16:
Still having this issue that the AP is open. The SSID is being showed now. But that mybe only works because I tried to create an open AP by purpose as I omitted the password argument. Now after I added the password again it shows the AP but still as open and not secured :/

I have version 2.3.0 through board manager

my Sketch:

`#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <WiFiUdp.h>

/* Set these to your desired credentials. */
const char *ssid = "artex11";
const char *password = "getme";
WiFiUDP udp;

void setup() {
	delay(4000);
	Serial.begin(115200);
	Serial.println();
	Serial.print("Configuring access point...");
	/* You can remove the password parameter if you want the AP to be open. */
	WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println("AP online!");

  Serial.println("Setting up UDP Listener...");
  udp.begin(11);
  Serial.printf("UPD listeing on %s:%d\n", WiFi.softAPIP().toString().c_str(), 11);
  

}

void loop() {
  int packet = udp.parsePacket();
  if(packet){
    Serial.printf("got one...");
  }

}`