esp8266/Arduino #1262: FSBrowser.ino Reboot

roboticboyer, 2015-12-20:
I'm testing: 
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino

I'm using an ESP-01 with 1kb of SPI memory (black pcb)
I haven't changed any part of the code.

The sketch is compiled without any error (arduino IDE 1.6.5 and ESP 2.0.0) but every 2,3 minutes the ESP crashes and reboots.
The error is:
>
Fatal exception (0):
epc1=0x40106556, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Exception (0):
epc1=0x40106556 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
ctx: sys 
sp: 3ffffd20 end: 3fffffb0 offset: 01a0
stack
3ffffec0:  4020efc9 00000023 00000000 3fffdb20  
3ffffed0:  3fff1fd8 3fff2dd8 3ffea8ec 4020f1da  
3ffffee0:  ffffffff 3fffc6fc 00000001 3fffdab0  
3ffffef0:  3fff13e4 3fffdc20 3ffefe04 00000030  
3fffff00:  3fff0000 40210242 3ffed4c8 3fff1568  
3fffff10:  3ffea8d8 3fff156c 3fff2dd8 4020f704  
3fffff20:  00000002 3fff1fd8 3fff1fd8 3ffeec80  
3fffff30:  00000000 3fff2dd8 0000001c 3fff1fd8  
3fffff40:  3ffea8ca 00000000 3fff2dd8 40215df5  
3fffff50:  6401a8c0 00000050 00000000 0000001c  
3fffff60:  00000002 00000018 4021d289 3ffed4c8  
3fffff70:  3ffea8a4 000000f5 3ffe9aa0 3ffe9aa0  
3fffff80:  4021d225 3ffed4c8 00000000 3fff2060  
3fffff90:  3fffdc80 00000000 3fff2dd8 40216847  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
stack
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42 

The ESP doesn't reboot with other sketches.

danielmawhirter, 2015-12-20:
I haven't seen this issue before with the same sketch, same board, same software versions. Any chance you've selected a bad/insufficient SPIFFS size? What operations do you/can you perform in the 2-3 minutes before it resets?

me-no-dev, 2015-12-22:
@igrr which lwip is in 2.0.0?

roboticboyer, 2015-12-22:
@danielmawhirter
The server is working but periodically it crashes and reboots. After rebooted it works again.
It's like a watchdog timer that reboots the ESP.

roboticboyer, 2015-12-23:
Now the sketch is working but I deleted the below rows of the code
`#include <ESP8266mDNS.h>`
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino#L27

`MDNS.begin(host);`
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/FSBrowser/FSBrowser.ino#L193

The issue seams to be related to mDNS library

roboticboyer, 2015-12-27:
There isn't any issue related to 3.3V power supply. It's OK.

me-no-dev, 2015-12-28:
the problem is not in mDNS but rather in the network stack. mDNS has been working really stable for quite a while now and code has not been changed really. 2.0.0 stable comes with build-from-source lwip which is the cause of most network troubles. I switched back to stock lwip from sdk 1.3 and all my issues disappeared.

roboticboyer, 2015-12-30:
@me-no-dev
How I can switch back to sdk 1.3?

liquidfalcon, 2016-01-21:
@me-no-dev Are you positive the 1.3 SDK works with mDNS? I've tried reverting all libs to 1.3, 1.4, and 1.5 and saw the exact same issue each time. Issue #1050 is the same as this too.

Has there been any development on this issue? mDNS / OTA seems to be completely broken in the newest staging, with the below stacktrace printed 10 seconds - 1 minute after starting mDNS.begin();

Exception (0):
epc1=0x40106d4e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff20:  4021f10b 077fcf5c 40214c34 3ffece00  
3fffff30:  3fff5750 00000001 4021f14a 40214c4d  
3fffff40:  4021ef59 3fff57c8 3fffff80 3fff41e0  
3fffff50:  3ffe0000 3fff5750 3ffee8b0 4021f90c  
3fffff60:  3fff57c8 3fff41e0 3ffebbca 4021e6ab  
3fffff70:  3fff41e0 00000014 4021ec56 3fff57c8  
3fffff80:  3fff41e0 3fffdc80 3fff4250 3fffdcb0  
3fffff90:  4022ae7f 3fff57c8 00000000 4020aac3  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

me-no-dev, 2016-01-21:
I have been now on 1.5.1 for a few days (the git master of this repo) and can say that the network is considerably faster. I have not had any sudden crashes or anything.
As for mDNS, I am running it all of the time + OTA and usually a web server on top. I have not experienced issues on 1.3 (1.3 branch in my fork) nor on 1.5.1 here.

liquidfalcon, 2016-01-21:
Curious then - I tried using the mDNS example on both the latest master of yours and this repository, and still see exceptions after ~10 seconds. Do you see any significant multicast traffic on your network? If I add logging to the mDNS parse packet function, I see a considerable amount of packets per second before it crashes - Perhaps this is the difference? Although I can't seem to understand how it'd crash the system context because of that. Probably need to fire up GDB and figure out what line it's dying on.

me-no-dev, 2016-01-21:
What you explain is really strange. I think that I have a pretty large network for home network, with lots of iDevices that use mDNS for everything and I run for weeks without issues.
when you get a stack trace like above, you can use xtensa-lx106-elf-addr2line <firmware.elf> and then enter the exception addresses you see that start with 402 and get where is the exception and how it got there :)

liquidfalcon, 2016-01-21:
Cheers @me-no-dev - Didn't even realise that handy tool was included. So, the exception shows it happened at lib1funcs.S:696, and the only location addr2line seems to be able to decode is core_esp8266_main.cpp:43, which would be atexit. Now, with some google-fu, this appears to be a divide by zero error. However, there's no division going on in the mDNS code, so, I think this'll take some more investigation..

me-no-dev, 2016-01-21:
hmm... looking at the trace more addresses starting with 402 should have decoded :( sad. See if you can get more info maybe on next fail... there must be a pattern. The mDNS on it's own is not a big deal as a service.

liquidfalcon, 2016-01-21:
I would've thought so too - The stack frames GDB sees are two at 0x4021f252, which then call the __umodsi3 modulus function, with nothing else. Its quite consistent though, I've never seen it print anything else - Same values every time. I figure without having the source for the libraries

igrr, 2016-05-07:
Should be fixed in git version. If you're still having issues, please follow up in #1826.

closed 2016-05-07 by igrr

joaofl, 2016-08-19:
Had very similar error out of nowhere. Took me some time to realize that it was caused by the absence of one of the sensors on the I2C bus. It seems that it was the I2C library the responsible.

Added two "if" and problem solved