esp8266/Arduino #1263: Access DHCP options in STA mode?

penfold42, 2015-12-20:
Can I access the additional DHCP options that my DHCP server is supplying to client ?

Things like sys log server can be provided to the clients and I'd really like to be able to access them.

Thanks !

igrr, 2015-12-22:
Whatever is implemented in LwIP, we have access to.
You can check dhcp client source code here: https://github.com/kadamski/esp-lwip/blob/esp8266-1.4.1/src/core/dhcp.c, or a slightly more up-to-date version at bbs.espressif.com (look for "open source LwIP for SDK 1.5").

penfold42, 2015-12-22:
Thanks - looks like it only parses DHCP options it really needs.

If anyone else is interested, I'm looking to extend it to  over things like syslog, ntp and so forth.

far5893, 2016-04-17:
I m interested

sauttefk, 2016-05-01:
I would like to have access to DHCP option 42 (NTP server). Unfortunately LWIP discards this option currently.

But as the code in /tools/sdk/lwip/src/ is apparently not being compiled in the Arduino IDE.
* How can I make changes to /tools/sdk/lwip/src/core/dhcp.c so they are being used in my projects?
* or even make them available to other people?

idolpx, 2016-05-17:
I'm interested as well.  I'd like to be able to set the The Captive-Portal DHCPv4 option (160).

samacumen, 2017-08-10:
Can esp8266 be used to automatically connect to a router. The router is a dhcp server and I want esp8266 to act as a dhcp client.

Any suggestions on how to do this and if this is possible?

Thanks in advance.

devyte, 2017-08-10:
Yes, it's standard usage. Study the examples included with the core, they
are accessible from the arduino ide menu. Please close this issue, this is
an issue tracker, not a support forum.

samacumen, 2017-08-10:
@devyte, was that a response to my query? What is accessible from arduino ide menu, could you clarify? This issue is already closed. Thanks.

devyte, 2017-08-10:
@samacumen ah yes, I was responding via email, so I couldn't tell the issue is already closed.
In the arduino ide there is an Examples item, where you can browse all included examples. When you click on one, it opens another instance of the ide, with the example opened. Just look through them, you'll find what you're looking for.

xuanduc611, 2017-09-20:
@samacumen Did you find the way to change between DHCP and StaticIP in Station mode?

devyte, 2017-09-20:
The examples are available from the Arduino IDE.
0.0.0.0 makes STA do dhcpc, anything else is interpreted as a static IP.
Please stop using this for questions that should be asked at a forum.

NarinLab, 2017-09-26:
Is there any clue on how to change DHCP Server default IP range from 192.168.244.x or 192.168.4.x to other custom range such as 192.168.43.x?

I can not find any documentation on this, I need to change the default DHCP Server options on AP Mode.

Please help..

NiKiZe, 2017-12-21:
I would like to see this reopened (unless there is some other issue or implementation for this already)
This cold be handled by for example by having a callback function that could handle all options in user code.

d-a-v, 2017-12-21:
@NarinLab @NiKiZe please open new issues for dhcp-server feature request into lwip2 [repo](https://github.com/d-a-v/esp82xx-nonos-linklayer)

NiKiZe, 2017-12-21:
@d-a-v what I'm looking for in this case is dhcp-client hook to get these options when connection to a network that might have them. (so not dhcp-server mode when ESP is running in AP mode,)

The reason for mention that here is because OP asked about the use case I'm after so it feels wrong to open duplicate issues for this. But yes, i will try to do so in the correct repo, thanks (unless there already exists).

To clearify: the usecase is to get things like syslog ip, or ntp ip in ESP client mode.
My suggestion is that this should be done by calling a user provided hook function/delegate since that can give any dhcp option and not just the ones mentioned above.

One thing to note tho is that the dhcp-client code also needs to know that these options should to be requested from dhcp-server (some dhcp servers send all options regardless, while others only sends what was asked from it)

NiKiZe, 2017-12-21:
It seems that @NarinLab s issue was solved in #1956

d-a-v, 2017-12-21:
@NiKiZe Thanks.

About your request, lwip2 (through lwIP-v2) already takes care of NTP servers in current master. Check the example esp8266/NTP-TZ-DST.

lwIP-v2 does not take care of option 7 syslog IP.
list of options currently taken care of: [link](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/include/lwip/prot/dhcp.h?h=STABLE-2_0_3_RELEASE&id=92f23d6ca0971a32f2085b9480e738d34174417b#n140)
switch-case for parsed option: [link](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/ipv4/dhcp.c?h=STABLE-2_0_3_RELEASE&id=92f23d6ca0971a32f2085b9480e738d34174417b#n1580)

We could make a compile-time-patch that would put there a user callback for unparsed options.

NiKiZe, 2017-12-22:
@d-a-v ntp and syslog are just examples, again it should be possible to get _any_ dhcp options.
So that callback should be for _all_ options regardless of if they are handled or not.
This can even be runtime, static var that can be set to pointer of user provided function, for each option check if (var != null) * var(option, data) (my C skills are horrible, but I know the technique to be good)

If you only send thru unparsed options you will risk breaking user code if any new options are implemented. So again send on _all_ options parsed or not.

I have rolled quite a few NTP solutions on my own since it wasn't available, now the question is how does one get the ntp server provided. (the above approach would solve that as well, and in a generic manner so I don't have to read sources for every option that I want to know about)

Also it is very confusing that you link to your private lwip2  repo, shouldnt that be under the esp8266 org?

devyte, 2017-12-22:
@NiKiZe lwip2 is currently experimental, and @d-a-v is handling the implementation pretty much as a one-man army. That's why the odd linking. Lwip2 won't become default for a while to come, but it's being made available because it is known to fix several issues to some specific users.
Your request for callbacks may be considered, but be aware that t won't be a priority in the near future, as we're focusing on progress for the 2.4 release.
I'll reopen this as enhancement in the meantime for further consideration later on.

d-a-v, 2017-12-22:
@NiKiZe 

I think the easier is to make a callback, either for all dhcp options or only the unhandled ones. The live patch at compile time would be straightforward. 
As I said NTP is now available and already efficiently handled by sntp in lwip in current master  (just check the example esp8266/NTP-TZ-DST)
And with PR #4001 even accurate microseconds will be set in ```gettimeofday()```. A user callback will be available and called as soon as time is set following dhcp answer.

About the lwip2 repo, it is currently linked as esp8266/Arduino's  [submodule](https://github.com/esp8266/Arduino/tree/master/tools/sdk/lwip2) and selectable in menus.

PR #6680 (2020-12-22): more lwIP physical interfaces
19-10-28:

* Ethernet: W5100, W5500 and ENC28J60, thanks to quality drivers from @njh repositories
https://github.com/njh/EtherSia/tree/master/src/enc28j60.cpp
https://github.com/njh/W5100MacRaw
https://github.com/njh/W5500MacRaw
SPI needs to be initialized before using these drivers.

* PPP server with example

* DHCP server moved out from lwip2 directories is now a per-interface instanciable class
  (and more easily improvable)

* Every services / libraries should work with ethernet 
  ~~- mDNS is [a work in progress](https://github.com/Labor-Et-Ars/ESP8266-Ardu [... cut]
closed 2020-12-22 by d-a-v