esp8266/Arduino #1264: Adding supported firmware

delcom5, 2015-12-21:
It would be nice to see SUPPORTED FIRMWARE listed on the Read me page.

penfold42, 2015-12-21:
I'm not sure that's relevant.

When you upload your code, it replaces any existing espressif firmware that was on the device

delcom5, 2015-12-21:
After uploading the simple blink nothing happened. Then when I tried to send simple AT commands and got no response. After research I flashed the firmware. After that I was able to send AT commands with success. I uploaded the simple blink and it finally worked...which is why I thought it could be a firmware issue.

itsjustvenky, 2015-12-21:
@delcom5  AT firmware is a app and blink is also an app. AT firmware is developed by espresiff itself and blink is done by people like yourself. What firmware version has to do here. I suggest you read more about ESP development.

pgollor, 2015-12-21:
Any other app (AT commands app or so) could overwrite some other registers.
It will be the same if you clear the spi memory and flash your blink app after that. There is no correlation between the firmware apps and the arduino apps.

Therefore i will close this issue.

closed 2015-12-21 by pgollor