esp8266/Arduino #1265: Maximum frequency PWM

sabas1080, 2015-12-21:
Hi

What is the maximum frequency of the PWM ESP8266 ?

Thanks

me-no-dev, 2015-12-21:
theoretical max frequency is 80000000/range, or for default 1024 range that is 78.125KHz in reality, depending on the number channels and the values that are given, the resolution and timing at that speed might be slightly off.
I'll make a pull request with some optimizations :) looking at the code, there are some that can be done.

sabas1080, 2015-12-21:
Thanks @me-no-dev, I'll try a 1Mhz

closed 2015-12-21 by sabas1080

mottab, 2016-07-14:
So, can I reach (32.768Khz) for master clock in PWM ??

Thanks.

tablatronix, 2017-03-14:
Care to add some references to this? Technical docs note 

> the function of PWM is realized by interruption of the
timer, the minimum resolution reaches as much as 44 ns. PWM frequency range is
adjustable from 1000 μs to 10000 μs, i.e., between 100Hz and 1 kHz. When the PWM
frequency is 1 kHz, the duty ratio will be 1/22727

drelephant, 2018-04-04:
From [here](http://esp8266.github.io/Arduino/versions/2.0.0/doc/reference.html#analog-output):

> analogWrite(pin, value) enables software PWM on the given pin. PWM may be used on pins 0 to 16. Call analogWrite(pin, 0) to disable PWM on the pin. value may be in range from 0 to PWMRANGE, which is equal to 1023 by default. PWM range may be changed by calling analogWriteRange(new_range).
>
> PWM frequency is 1kHz by default. Call analogWriteFreq(new_frequency) to change the frequency.

So I think all you have to do is call analogWriteFreq(32768) to run at 32.768kHz.

diraniyoussef, 2020-01-22:
[Here](https://github.com/esp8266/Arduino/blob/master/doc/reference.rst) they made it clear, check "Analog output" section. It's 40 KHz the maximum.

quetzalcoatl, 2021-02-17:
@me-no-dev you've pointed out that the theoretical max is:
> 80000000/range
I can't help but notice that's the 80MHz clock rate. I suppose it's only achievable with software and ~50% duty cycle.
That should mean that when ESP8266 is switched to 160MHz, it's theor.max. 160000000/range?
Could you confirm?

me-no-dev, 2021-02-18:
@quetzalcoatl no. peripherals run from the APB clock which is always 80MHz. Same is true for ESP32