esp8266/Arduino #1268: client.connected() fails when IPaddress is mentioned

suranaumang, 2015-12-21:
Hi,

I am new to esp8266 and trying to connect it with PC using locally hosted mosquitto server. When the mqtt server's domain name is mentioned the esp connects but when it is changed to IPAddress of the domain, mqtt connection fail. I am working on mqtt_esp8266 example in Pubsub client. 
The only change I am making is to the following line-

const char* mqtt_server = "broker.mqtt-dashboard.com";

changing it to

IPAddress mqtt_server = (212,72,74,21);   // the ip address of the above mentioned link

This shows the error=-2 i.e. mqtt connection failed.

pgollor, 2015-12-21:
Which mqtt library do you mean? I think there is no mqtt library inside the Arduino ESP8266 package.

itsjustvenky, 2015-12-21:
@suranaumang "broker.mqtt-dashboard.com" and "212,72,74,21" are not same.

video.youtube.com != youtube.com (xx.xx.xx.xx)

devyte, 2017-10-19:
Closing per #3655 .

closed 2017-10-19 by devyte