esp8266/Arduino #1274: streaming writes to ESP8266HTTPClient over SSL

gtalusan, 2015-12-21:
I have a plain old HTTP response stream (WiFiClient *) that I'm passing to HTTPClient::sendRequest(method, Stream *, size_t) over SSL.

The while-loop for reading from the stream and writing to the _tcp instance only does one iteration.
```
    while(connected() && stream->available() && (len > 0 || len == -1)) {
```
shouldn't check stream->available() since it could return 0 over a slow-ish connection.  Maybe check for -1?

Similarly, the call here:
```
           // write it to Stream
            bytesWritten += _tcp->write((const uint8_t *)buff, c);
```
is returning -1 when _tcp is a WifiClientSecure.

```
size_t WiFiClientSecure::write(const uint8_t *buf, size_t size) {
    if (!_ssl)
        return 0;

    int rc = ssl_write(*_ssl, buf, size);
    if (rc >= 0)
        return rc;

    return 0;
}
```
The last return should maybe be a -1 with errno set, or simply just returning rc.

I'm not familiar with the Arduino API enough to submit a patch for these issues.

Links2004, 2015-12-22:
the available part is fixed.

but i dont unterstand the ssl write problem.

size_t == uint32_t on the ESP which mean no -1.
and the ssl_write will return the written bytes:
https://github.com/igrr/axtls-8266/blob/514b6685c5a84232caeed72dc5720c652bbe9f73/ssl/tls1.c#L304

can you explain it more?

igrr, 2015-12-22:
ssl_write can return an error code as well

Links2004, 2015-12-22:
yes and on error the return of  WiFiClientSecure::write is 0 which is good from my point of view.

gtalusan, 2015-12-22:
Thanks for fixing the part about available!

I see your point about size_t.  But there are more problems with the code you pointed out here:

https://github.com/igrr/axtls-8266/blob/514b6685c5a84232caeed72dc5720c652bbe9f73/ssl/tls1.c#L304

ssl_write will call send_packet and that can return an int.  In certain cases it'll return the value of SSL_ERROR_CONN_LOST which is a negative number and ssl_write will return this, but write() will hide this fact.

I see from Print.h there is setWriteError that can set the state of the stream internally.  Do you think  HTTPClient can use this to better know when the streaming loop should continue or not?

Links2004, 2015-12-22:
i can add getWriteError as Abort case. 
does WiFiClientSecure set it?

gtalusan, 2015-12-23:
I submitted a pull request here: #1289 to gracefully recover from a short write.  In the case of a ssl_write returning 0 all the time, the loop would just go on forever until the underlying HTTP connection closed or the watchdog killed me.

The code also contains a fix to StreamString that was obvious when the short write handling was fixed.

I've worked around my short write SSL problem by writing to SPIFFS.  Previously I was trying to stream from a plain text WifiClient to a WifiSecureClient.  I may try that again eventually, but for now #1289 gets me on my way.

closed 2015-12-27 by Links2004