esp8266/Arduino #1275: Exception (28)

svdgraaf, 2015-12-21:
I'm running into an issue, but I don't know where I can find the type of exception I'm getting:

```
Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1310 end: 3fff1540 offset: 01a0

>>>stack>>>
3fff14b0:  3fff1510 3ffe908c 3fff029c 402036f7  
3fff14c0:  3fff1510 0000002b 3ffe8f1e 3fff0518  
3fff14d0:  3fff1510 3ffe908c 3fff029c 40203b81  
3fff14e0:  3fffdad0 00000000 3fff029c 40203d08  
3fff14f0:  3ffe8d4d 00000000 3fff04e4 40203e65  
3fff1500:  3ffe8374 00000000 3fff04e4 40202284  
3fff1510:  3fff1ec0 27000000 3fff1e98 00000000  
3fff1520:  3fffdad0 00000000 3fff0511 40205d70  
3fff1530:  00000000 00000000 3fff0520 4010011c  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
I saw some issues mentioning memory usage, but I'm not sure if that's the case here.
```
Sketch uses 280,522 bytes (26%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 38,260 bytes (46%) of dynamic memory, leaving 43,660 bytes for local variables. Maximum is 81,920 bytes.```

Can someone give me any pointers? Where could I find a list of the exceptions that there are? Where can I read some more? :)

igrr, 2015-12-22:
Check out [exception causes list](https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md) and [GDBStub library](https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub) which may help you debug this issue.

svdgraaf, 2015-12-22:
Awesome, thanks! I'll take a look at it :)

svdgraaf, 2015-12-24:
AWESOME! I learned so much from this :)

I ended up making a stupid mistake. For people getting the same problem later on, follow the steps above. Once you're in `gdb`, you can get a stacktrace by typing in `backtrace` in the gdb console.

closed 2015-12-24 by svdgraaf

Testato, 2016-01-01:
@svdgraaf 
may you tell whitch error you found ?
I have this message that is quite similar to your

```
Exception (28):
epc1=0x402066e6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3ffeace0 end: 3ffeaeb0 offset: 01a0

>>>stack>>>
3ffeae80:  3fffdc20 3ffe9a68 3ffeaed4 40202a8e  
3ffeae90:  00000000 00000000 3ffeaed4 40201c0a  
3ffeaea0:  00000000 00000000 3ffe9e90 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

svdgraaf, 2016-01-01:
I had a loop somewhere where it shouldn't be ;) Just try to setup the debugger, you'll get a lot more information as a bonus :)

HW-Siew, 2016-03-18:
Hi @svdgraaf , would you mind sharing how you get the gdb lib setup?

svdgraaf, 2016-03-18:
@HW-Siew check out the description on the GDBStub page: https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub

You have to include the library (download and put it somewhere in the Arduino path), run the python tool (tcp_serial_redirect.py), leave it running, and connect to it with the `nc` command.

When the software on your esp crashes, exit the `nc` command (or leave it running, that's fine), and use the gdb tool to connect to the error process.

You can get a backtrace on the `gdb` prompt, by doing `backtrace`, which would give you a backtrace of all the calls that were done, before running into the issue. There are a lot more options available in the gdb console, but the `backtrace` is all I needed to find my bug, I think you can get a list of all commands by running `help`.

aliceand, 2017-02-08:
I'd like to use GDBSub but I'm using Arduino IDE with my ESP8266.  The GDBStub setup looks like it requires Linux.  Is there a way I can use GDBStub with my Arduino IDE?

Adsuarez, 2017-12-09:
Hi @aliceand, how could you solve your problem? i have de same problem :(

d-a-v, 2018-01-08:
This is a closed issue. If you need help, follow the rule: read the doc (policy, troubleshooting) and:

Open a *new* issue with a minimal sketch reproducing the error.

brianjmurrell, 2018-02-03:
> Check out [exception causes list](https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md) and

That's a 404 now.  Seems it's now https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.rst.

coogle, 2018-05-03:
Can you please provide a sketch that causes this crash for you? I'd be happy to try it on the ESP chips I have to see if I can reproduce and fix it.

Leobute, 2018-06-30:
Go the same error. Reset to esp, I runs for 1 min morfe or less perfectly, then ….error.
Don't know how what to do anymore.

Exception (28):
epc1=0x4020398b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3ffefd60 end: 3ffeff60 offset: 01a0

>>>stack>>>
3ffeff00:  3ffeeca4 3ffeebf0 3ffeff20 3ffeebfc
3ffeff10:  3ffeeca4 3ffeebf0 3ffeec8c 4020254c
3ffeff20:  401070a0 b65a10ac 00000000 feefeffe
3ffeff30:  00000000 00000000 00000001 3ffeef30
3ffeff40:  3fffdad0 00000000 3ffeef28 40203d78
3ffeff50:  feefeffe feefeffe 3ffeef40 40100710

leifnel, 2018-12-24:
The command line for tcp_serial_redirect.py 
`./tcp_serial_redirect.py -p /dev/tty.SLAB_USBtoUART -b 115200 --spy -P 9980 --rts=0 --dtr=0`

does not work with the one I found at [https://github.com/pyserial/pyserial/blob/master/examples/tcp_serial_redirect.py](https://github.com/pyserial/pyserial/blob/master/examples/tcp_serial_redirect.py)

`
usage: tcp_serial_redirect.py [-h] [-q] [--develop] [--parity {N,E,O,S,M}]
                              [--rtscts] [--xonxoff] [--rts RTS] [--dtr DTR]
                              [-P LOCALPORT | -c HOST:PORT]
                              SERIALPORT [BAUDRATE]
tcp_serial_redirect.py: error: unrecognized arguments: -p -b 115200 --spy
`

Are there other like-named `tcp_serial_redirect.py` around?

sandevid, 2019-03-11:
### How to fix this , i use nodeMCU
**Error :**
Exception (28):
epc1=0x4020a5d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000001bb depc=0x00000000

stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 01a0
3fffff30: 3fffdad0 00000000 3ffeed38 4020261b
3fffff40: 000001bb 3ffeed44 3ffe8884 3ffeeedc
3fffff50: 000001bb 3ffeed44 3ffe8884 40204a42
3fffff60: 40207ec0 dca79a95 3ffeed44 40203cbf
3fffff70: 3fffdad0 3ffeee10 3ffeed38 402057f0
3fffff80: 3fffdad0 3ffeee10 3ffeee64 402025d5
3fffff90: 40207ec0 552ba8c0 feefeffe feefeffe
3fffffa0: feefeffe 00000000 3ffeeeac 40207550
3fffffb0: feefeffe feefeffe 3ffe8558 40100a0d
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

**Source Code :**

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <TelegramBot.h>

#define LED 14 //D3

const char* ssid = "Telegram";
const char* password = "getupdate";
const char BotToken[] = "660345429:ACGlSKhCRMmgKHNJH4pG3ut_1R_66flenqQ";

WiFiClientSecure net_ssl;
TelegramBot bot (BotToken, net_ssl);

void setup() {
Serial.begin(115200);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println("");
Serial.println("WiFi connected");
Serial.println(WiFi.localIP());

bot.begin();

pinMode(LED, OUTPUT);
}

void loop() {

message m = bot.getUpdates();

if (m.text.equals("on")){
digitalWrite(LED, HIGH);
bot.sendMessage(m.chat_id, "The Led 1 is now ON");

}else if (m.text.equals("off")){
digitalWrite(LED, LOW);
bot.sendMessage(m.chat_id, "The Led 1 is now OFF");
}
}`

Board : NodeMCU 1.0 (ESP-12E Module)
Upload Speed : 115200
Flash Size : 4M (no-SPIFFS)
Debug Port : Disabled
Debug level : None
IwIP Variant : v2 lower memoty
VTables : Flash
Exceptions : Disable
Erase Flash : Only Sketch
Port : COM4

Please help,
I have searched everywhere but there is no right solution

10xnorellana, 2019-07-29:
i started the connection to wifi again before to use the http.begin(). it worked for me

ethanlooney, 2020-05-22:
I am having the same issue as the above comments, has anyone been able to find a solution to this issue?