esp8266/Arduino #1278: [sdk 1.5] Wifi does not connect sometimes

vlast3k, 2015-12-22:
I have a very strange problem with SDK 1.5. And a standard Nodemcu v0.9 board
I am testing with the standard WifiClient example. 
In my case I enable it and disable it via the CH_PD from an arduino. (with the nodemcu board, simply connect one wire from the "EN" pin to ground for a second)
With SDK 1.5 (current) - sometimes maybe half of the times - after enabling it, the Wifi does not connect
With SDK 1.3 (latest change before SDK 1.5 was introduced) - it always connects
With the AT FW 0.51 (that came with SDK 1.5), it also connects each time.

I wonder if somebody else will be able to repeat this ?

me-no-dev, 2015-12-22:
same here. need to power cycle most of the time in order to connect to the WiFi.
some changes to the init were introduced today that might help you in getting better chance for connection. Pull the latest git and give it another go :)

windalser, 2015-12-27:
Try  
WiFi.disconnect(); 
somewhere inside setup() before any  client.connect(...)

vlast3k, 2015-12-29:
It took me some time to retest it, but now it seems to be working fine
So i am closing the issue

closed 2015-12-29 by vlast3k