esp8266/Arduino #1280: ESP8266 can't conect WEB, Please help me :(

hadinhdiep, 2015-12-22:
Hi all,
I'm a new ESP8266, I test WiFiClient.ion on my devices but  my esp8266 can't connect https for get data. This's code  // Use WiFiClient class to create TCP connection
                                            const int httpPort = 80;
                                            if (!client.connect(host, httpPort)) {Serial.println("connection failed");
    return; }
I dont know why, pleas tell me. Thank you verry much !!

Links2004, 2015-12-22:
for https you need port 443 and WiFiClientSecure.
or you use ESP8266HTTPClient
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino#L46

hadinhdiep, 2015-12-22:
I have tested this project https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient.ino
I replace my SSID, Password but can't connect server data.sparkun.com, 
"Connection failed".

hadinhdiep, 2015-12-22:
Thank you verry much. I'm done.!

closed 2015-12-23 by Links2004