esp8266/Arduino #1281: Clarification: use of hostname symbol?

andig, 2015-12-22:
When I declare

    String hostname;

I end up with a compiler conflict:

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc2/tools/sdk/lib\libmain.a(eagle_lwip_if.o):(.bss+0x0): multiple definition of `hostname'

	sketch\esp8266-webconf.ino.cpp.o:(.bss.hostname+0x0): first defined here

	c:/users/xx/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: Warning: size of symbol `hostname' changed from 12 in sketch\esp8266-webconf.ino.cpp.o to 4 in C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0-rc2/tools/sdk/lib\libmain.a(eagle_lwip_if.o)

	collect2.exe: error: ld returned 1 exit status

	exit status 1
	Error compiling.

However, when I try to use the obviously existing internal `hostname` I receive an undefined error.

Is `hostname` a forbidden symbol name?

igrr, 2015-12-22:
hostname is a symbol exported from one of SDK libraries. When you declare
another symbol with the same name, you get linker error. If you try to use
the existing one you get compiler error, because hostname isn't declared
anywhere.

For now, you should avoid using this name.
I'll check if it can be localized to one of the libraries.

devyte, 2017-10-19:
@d-a-v does #3362 fix this? I can't find the symbol in lwip2.

devyte, 2017-11-20:
@igrr said:

> I'll check if it can be localized to one of the libraries.

What does this mean? Did you ever get around to looking into it?

igrr, 2017-12-27:
hostname seems to be also defined in lwip2 now:
```
tools/sdk/lib/liblwip2.a:lwip-git.o:00000000 b hostname
```

devyte, 2017-12-28:
@igrr you know, this can be avoided with an anonymous namespace, which is recommended C++ style anyways (put user globals in an anonymous namespace in the .ino). Whatever else we do for this issue, that should be our recommendation for usage of globals anyways.

d-a-v, 2017-12-28:
@igrr hostname in lwip2 was not used and is not anymore present in sources.
The hostname defined in libmain.a is probably used by ```wifi_station_get/set_hostname()```.

devyte, 2017-12-28:
@d-a-v @igrr : 

```
develo@esp:~/workspace/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder$ nm liblwip2.a |grep hostname
00000000 b hostname
...
develo@esp:~/workspace/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder$ grep -r -n hostname .
...
./glue-lwip/lwip-git.c:50:static char hostname[32];
...
develo@esp:~/workspace/arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2$ nm /tmp/arduino_build_356920/espora.ino.elf  |grep hostname
...
3fff1740 b hostname
3ffefab4 B hostname
...
```

I looks like now we have 2 hostname symbols...

d-a-v, 2017-12-28:
After current lwip2 PR there will be only one.

devyte, 2017-12-28:
Ok, so we're back to the initial issue: the hostname symbol likely coming from the SDK.
@igrr said:

> I'll check if it can be localized to one of the libraries.

What can be done?

igrr, 2017-12-28:
Ok, if the hostname symbol does not have to be shared between libmain and lwip, I'll just make it local in the object file where it is defined.

igrr, 2017-12-31:
Resolved in https://github.com/esp8266/Arduino/commit/8edeac0cf4f99f45c275a41a927d1e6023f95398.

Localizing the symbols was not possible (they were used in two object files), so i added `wifi_station_` prefix to both `hostname` and `default_hostname`.

closed 2018-01-02 by igrr