esp8266/Arduino #1282: ArduinoIDE 1.6.6 doesn't find BasicOTA port on Windows7 x64

andig, 2015-12-22:
Following the OTA examples and install instructions. Sketch uploads fine, ESP is responding to mDNS. However, the OTA port doesn't show up (1.6.6 IDE, 2.0.0rc2 library).

Any hints where to look?

And- lest I forget- awesome project, thank you  so much!

me-no-dev, 2015-12-22:
1.6.6 is the only version that can not run OTA :) try 1.6.7

andig, 2015-12-22:
Same problem with 1.6.7.

```
Booting
OTA server at: esp8266-06ac87.local:8266
Ready
IP address: 192.168.0.71

Even though the sketch is running (see above) I cannot see the OTA port. Running upload via OTA without previously selecting the port consequently fails.

andig, 2015-12-22:
Hah! Problem solved. I had to disable "Auto" proxy settings in the IDE. No clue why board manager was fine, but now OTA is visible- thank you!

closed 2015-12-22 by andig

me-no-dev, 2015-12-22:
it's a known issue with Arduino IDE which fails to see the boards after it has been open for a while. Restarting the IDE helps