esp8266/Arduino #1283: No random() function

NicoHood, 2015-12-22:
Arduino uses `random` as identifier for a random function. However I just tested and rand() was available. Maybe someone should set a shortcut. It would be also interesting how relyable this random is (can it used for cryptography or better not)?

Links2004, 2015-12-22:
we have random:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Arduino.h#L267-L269

the question regarding cryptography its most depends on your seed.
the best seed src we have is this:

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/esp8266_peri.h#L837-L841

NicoHood, 2015-12-22:
random cannot be found within this program:
https://github.com/NicoHood/tweetnacl-usable/blob/arduino/randombytes.c#L11

Maybe because its .c not .cpp?

Links2004, 2015-12-22:
rand != random,

NicoHood, 2015-12-22:
Yes, if you replace it with random, it will not work anymore.

Also whats the difference between those functions?

Links2004, 2015-12-22:
rand is part of the cstdlib from C
http://www.cplusplus.com/reference/cstdlib/rand/
needs for C:
```c
#include <stdlib.h>  
```
or
CPP:
```cpp
extern "C" {
#include <stdlib.h>  
}
```
random is a API function from Arduino
https://www.arduino.cc/en/Reference/Random
not working on C

and needs for CPP 
```cpp
#include <Arduino.h>  
```

the ESP supports both.
rand and random

NicoHood, 2015-12-22:
Okay then it was my mistake, nevermind. Thx for the infos!

closed 2015-12-22 by NicoHood