esp8266/Arduino #1291: ESPhttpUpdate.update Magic byte is wrong

zenmanenergy, 2015-12-23:
I'm struggling to get the ESPhttpUpdate.update example working. I've edited the httpUpdate example to change the binary to this:

t_httpUpdate_return ret = ESPhttpUpdate.update("http://www.fridgealive.com/api/1.4/bin/helloworld_serial.cpp.bin");

It gives me this error on an esp-01:

[httpUpdate] Update.end failed! (ERROR[10]: Magic byte is wrong, not 0xE9)
[httpUpdate] Update failed
HTTP_UPDATE_FAILD Error (10): Update error: ERROR[10]: Magic byte is wrong, not 0xE9[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 225200
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 671744
[httpUpdate]  - current Sketch Size: 288020
[httpUpdate] runUpdate flash...

Links2004, 2015-12-23:
when i download the file manually the first byte is 0xE9.
the binary itself is ok and compiled for 1MiB flash.

possible reasons:
- corruption during transmission (send your server a MD5?)
- corruption during flash write
- flash is defektiv

baruch, 2016-03-25:
It can also be that the normal web browser and the esp http client do not behave exactly the same.

@zenmanenergy, you might want to dump the first 1K of the data received in the http client and see what is in there to try and decipher what went wrong.

I'm using the http update with a Python based server and it all works nicely without any issues.

closed 2017-05-15 by zenmanenergy