esp8266/Arduino #1293: softAPIP 0.0.0.0, race condition ?

tablatronix, 2015-12-23:
I was having issues with wifimanager not starting my ap up when I had a delay of 200ms.

Was thinking maybe some race against SDK auto connect, can't seem to find that documentation though... have not checked source.

```
#include <ESP8266WiFi.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.disconnect();

  // TEST
  // WiFi.mode(WIFI_OFF);  // workaround BREAK FIX
  delay(200); //this delay (>~100ms), if you comment out it all works, if left, softAPIP returns 0.0.0.0 BREAK
  // END TEST
  
  WiFi.mode(WIFI_STA);
  WiFi.begin("", "");
  int connRes = WiFi.waitForConnectResult();
  WiFi.mode(WIFI_AP);
  //assuming connecting failed
  WiFi.softAP("asdfghj");
  delay(500); 
  Serial.println("");
  Serial.println(WiFi.softAPIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

tablatronix, 2015-12-23:
ref
https://github.com/tzapu/WiFiManager/issues/31

Links2004, 2015-12-30:
can you retest it with the latest git?
have done many changes to the WiFi class

tablatronix, 2015-12-30:
Will do

tablatronix, 2016-01-02:
No change

tzapu, 2016-01-09:
this seems sorted with the latest github changes and sdk 1.5.1
give it a try if you can

Duality4Y, 2016-01-09:
This happens to me often using sdk 1.5.0

Links2004, 2016-01-09:
check latest git with SDK 1.5.1

Duality4Y, 2016-01-09:
seems to fix it here. tried it a few times, and it hasn't given me the 0.0.0.0 yet!

closed 2016-02-29 by igrr