esp8266/Arduino #1297: Make it possible to use ArduinoOTA without mDNS?

sven337, 2015-12-24:
As it is currently implemented,  the  ArduinoOTA class:
1-requires #include <ESP8266mDNS.h> *in the sketch*(I'm not sure why its own include seems not to work)
2-effectively uses mDNS with no ability to disable it

This issue is a request to make it possible to use ArduinoOTA without mDNS. The justification for me is that I don't want mDNS in the first place, that it's not required for OTAs, and that it bloats the sketch size (which on small flash sizes is a real issue).

Something like #define ARDUINO_OTA_NO_MDNS would probably be feasible?

Thank you

igrr, 2015-12-24:
1. Include of mDNS library in the sketch is required on Arduino 1.6.5/1.6.6, but shouldn't be required on 1.6.7. This is because library dependency tracking has been added to arduino-builder.
2. Will look into this

me-no-dev, 2015-12-24:
ArduinoOTA without the mDNS backing is just OTA. It's painless to add a parameter to the begin method to not start mDNS.
That would not fix the "missing" include, but as user move on to newer versions of the IDE, it will fade away.

DaAwesomeP, 2016-12-24:
Is it possible to manually specify an IP in the IDE for OTA instead of using mDNS?

me-no-dev, 2016-12-25:
the IDE is using mDNS to discover the devices, so no you can not, but you can run the command manually and upload your bin to IP (you would need to comment [those](https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/ArduinoOTA.cpp#L129-L135) lines to stop mDNS)

DaAwesomeP, 2016-12-25:
I settled on using ESP8266HTTPUpdateServer. Thanks!

download13, 2018-11-05:
This would be really useful for my case as well. I'm using another library that already uses mDNS for auto-discovery. It seems redundant to have it broadcasting two different names for no reason. Especially since I'm using a command line tool and don't need to use the IDE to send OTA updates.

devyte, 2018-12-14:
Closing via #5494 
ArduinoOTA.begin()',now accepts an optional bool flag, which when false skips manipulation of MDNS.

closed 2018-12-14 by devyte