esp8266/Arduino #1299: WiFi connection to non-existent AP failes with stack dump

tbnobody, 2015-12-24:
I successfully connected the ESP to my WiFi. If I disconnect the AP or start the ESP out of range of the AP the ESP crashed with a stack dump. 
The sketch is very simple:
```C
#include <ESP8266WiFi.h>
#define BAUD_RATE 9600

void setup()
{

	// Initialize serial output
	Serial.begin(BAUD_RATE);
	while (!Serial) yield();
	Serial.println(F("Starting ESP8266"));

	Serial.println(ESP.getResetInfo());

	WiFi.begin();
	while (WiFi.status() != WL_CONNECTED) {
		delay(500);
		Serial.print(".");
	}
	Serial.println("");
	Serial.println("WiFi connected");

}

void loop()
{
}
```

and the output is:
```
Starting ESP8266
Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
.......................................................................................................................................................................................................................
Exception (29):
epc1=0x402242ff epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000038 depc=0x00000000

ctx: sys 
sp: 3ffffd90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff30:  00000001 00000000 402189e3 3ffe9704  
3fffff40:  40219644 00000001 3fff2050 00000000  
3fffff50:  00000010 3ffe9704 3ffe9070 4021931a  
3fffff60:  3ffed3c8 3ffea7a4 000000f5 3ffe99c8  
3fffff70:  3ffed3c8 3ffea7a4 000000f5 3ffe99c8  
3fffff80:  00000040 3ffed3c8 00000000 3fffdcb0  
3fffff90:  40218ca0 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffe99c8 40000f49 3fffdab0 40000f49  
<<<stack<<<
```
The stack dump is not always exactly the same. Here another example
```
Starting ESP8266
Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
...............................................................................................................................................................................................................................................................................................................................................................
Exception (29):
epc1=0x402242ff epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000038 depc=0x00000000

ctx: sys 
sp: 3ffffd90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff30:  00000001 00000000 402189e3 3ffe9704  
3fffff40:  40219644 00000001 3fff2050 00000000  
3fffff50:  0000002f 3ffe9704 3ffe9070 4021931a  
3fffff60:  3ffed3c8 3ffeb42c 000000f5 3ffe9920  
3fffff70:  3ffed3c8 3ffeb42c 000000f5 3ffe9920  
3fffff80:  00000040 3ffed3c8 00000000 3fffdcb0  
3fffff90:  40218ca0 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffe9920 40000f49 3fffdab0 40000f49  
<<<stack<<<
```
As soon as I power on the AP everything works very fine and reliable.

tbnobody, 2015-12-24:
When providing the credentials to the begin() method, I am also getting the following errors:
```
Starting ESP8266
Fatal exception:4 flag:1 (WDT) epc1:0x40000f68 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
....
Exception (0):
epc1=0x4021dfe1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffde0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff80:  00000040 3ffecc98 00000000 3fffdcb0  
3fffff90:  402129d4 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffe9228 40000f49 3fffdab0 40000f49  
<<<stack<<<
```
It seems that for some reasons the watchdog gets triggered.

tbnobody, 2015-12-24:
I am not sure whether it's a really issue... Because after several times calling ESP.eraseConfig(); it seems to work again. (not 100% reliable but it doesn't crash so often)

tbnobody, 2015-12-25:
After several times just disconnecting the AP from power the ESP crashes again with many different errors. Sometimes the errors occur directly after restart, sometimes some seconds after restart. Found out, that it also happens when changing the WiFi mode from AP+STA to AP. Is there somewhere a documentation which explains the different error codes?

Just a view examples which I was able to get just after pulling the power plug (just removed the stack dump):
```
Fatal exception (0):	epc1=0x4020faa0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000002d, depc=0x00000000
Fatal exception (0):	epc1=0x4020faa0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4020face, depc=0x00000000
Fatal exception (0):	epc1=0x4021e824, epc2=0x00000000, epc3=0x00000000, excvaddr
Fatal exception (2):	epc1=0x3fff24d8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3fff24d8, depc=0x00000000
Fatal exception (29):	epc1=0x4022cfbf, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000038, depc=0x00000000
Fatal exception (3):	epc1=0x40221f5f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4022197c, depc=0x00000000
Fatal exception (9):	epc1=0x4022cfbf, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000002d, depc=0x00000000
```

Links2004, 2015-12-25:
the Fatal exception are here:
https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md

the rest of the message are memory addresses and point to "code"
you can use
```
xtensa-lx106-elf-objdump -S your_elf_file.elf > dum.dobj
```
to get the function name to the addresses.
you need to use the elf that match the build (bin file) where the errors are outputted or the addresses are may pointing to other code.

tbnobody, 2015-12-25:
Thank you for the hints. I recompiled the bin/elf file, uploaded it again, just to be sure the object file and the code on the flash are the same. Than I unplugged the power from the AP.
```
beacon timeout
state: 5 -> 0 (1)
rm 0
pm close 7 0 0/26595053
f -180, WiFi disconnection detected.
scandone
no xxxxxx.xxx found, reconnect after 1s
reconnect
f 0, scandone
no xxxxxx.xxx found, reconnect after 1s
reconnect
f -180, scandone
no xxxxxx.xxx found, reconnect after 1s
reconnect
f 0, Fatal exception (0):
epc1=0x4020fa61, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x4020fa61 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff90:  00000000 3fffdc20 3fff09f8 4020fa4b  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
c_Çc_ÇÏ<Ÿ‚‚Žû
```

The dumped object file around this address shows the following:
```
4020fa28:	f0c112        	addi	a1, a1, -16
4020fa2b:	036102        	s32i	a0, a1, 12
4020fa2e:	0261c2        	s32i	a12, a1, 8
4020fa31:	c60b01        	l32r	a0, 40201260 <delay_end+0x18>
4020fa34:	0000c0        	callx0	a0
4020fa37:	fff631        	l32r	a3, 4020fa10 <_ZSt25__throw_bad_function_callv+0x10>
4020fa3a:	c597c1        	l32r	a12, 40201098 <uart_write_char_d+0x78>
4020fa3d:	006322        	s32i	a2, a3, 0
4020fa40:	fff531        	l32r	a3, 4020fa14 <_ZSt25__throw_bad_function_callv+0x14>
4020fa43:	0c2d      	mov.n	a2, a12
4020fa45:	fff601        	l32r	a0, 4020fa20 <_ZSt25__throw_bad_function_callv+0x20>
4020fa48:	0000c0        	callx0	a0
4020fa4b:	0c2d      	mov.n	a2, a12
4020fa4d:	fff501        	l32r	a0, 4020fa24 <_ZSt25__throw_bad_function_callv+0x24>
4020fa50:	0000c0        	callx0	a0
4020fa53:	a28c      	beqz.n	a2, 4020fa61 <_ZL9loop_taskP11ETSEventTag+0x39>
4020fa55:	fff021        	l32r	a2, 4020fa18 <_ZSt25__throw_bad_function_callv+0x18>
4020fa58:	fff141        	l32r	a4, 4020fa1c <_ZSt25__throw_bad_function_callv+0x1c>
4020fa5b:	72a032        	movi	a3, 114
4020fa5e:	f17d05        	call0	40201230 <__panic_func>
4020fa61:	3108      	l32i.n	a0, a1, 12
4020fa63:	21c8      	l32i.n	a12, a1, 8
4020fa65:	10c112        	addi	a1, a1, 16
4020fa68:	f00d      	ret.n
4020fa6a:	7c0000        	excw
4020fa6d:	1069      	s32i.n	a6, a0, 4
4020fa6f:	00b840        	excw
4020fa72:	124010        	excw
```

Than, just after the restart of the ESP it crashed again. The C Source looks like that:
```C
void setup()
{
	// Initialize serial output
	Serial.begin(BAUD_RATE);
	while (!Serial) yield();
	Serial.println(F("Starting ESP8266"));
	Serial.setDebugOutput(true);

	// Initialize file system
	if (!SPIFFS.begin()) {
		Serial.println(F("Failed to mount file system"));
	}

```
The serial output was as followed:
```
‡úÿc_ÇrSöfêStarting ESP8266
Fatal exception (0):
epc1=0x40100003, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x40100003 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1640 end: 3fff1a20 offset: 01a0

>>>stack>>>
3fff17e0:  4020fcbe 00158000 00000100 4020fb48  
3fff17f0:  0015a100 0015a000 0015a000 40211124  
3fff1800:  0000001a 00000018 04000102 3fff1aa8  
3fff1810:  3fff3130 00000100 3fff0a00 003f20fe  
3fff1820:  3fff1aa8 3fff3108 0015a000 3fff311c  
3fff1830:  3fff1aa8 3fff3108 0015a000 4020580d  
3fff1840:  00000100 3fff2f30 00000000 00000100  
3fff1850:  003f8100 003f80fe 003f8100 00000100  
3fff1860:  3fff1aa8 0005a000 0000002d 4020348d  
3fff1870:  00000000 00000080 00000000 000028b0  
3fff1880:  00000000 000028af 40203128 00000000  
3fff1890:  3fff2f30 00000000 00000020 00000100  
3fff18a0:  0000001f 00000000 00000000 00000015  
3fff18b0:  00000000 00000000 00000000 00000000  
3fff18c0:  00000001 00000000 3fff1aa8 40203860  
3fff18d0:  00000000 00000000 3fff18e6 3fff18e0  
3fff18e0:  0000001e 002c0429 00000000 00000000  
3fff18f0:  00100000 00002000 00100000 00000100  
3fff1900:  00000000 00000000 00000000 00000020  
3fff1910:  00000001 000000fc 3fff1aa8 40202330  
3fff1920:  3fff1960 00000000 3fff2e30 3fff3108  
3fff1930:  3fff30f8 4020f9ae 3fff1aa4 00000578  
3fff1940:  000000b4 3fff1aa8 3fff1aa4 40210ecd  
3fff1950:  00000578 40211b40 00000000 00000000  
3fff1960:  402110a8 4021116c 402112b0 002fb000  
3fff1970:  00100000 00001000 00002000 00000100  
3fff1980:  00000200 00002fb0 3fff1a70 402106b1  
3fff1990:  3ffe864c 00000001 00000001 3fff09f8  
3fff19a0:  3fffdc20 3fff1aa8 3fff1aa4 40210f17  
3fff19b0:  3fffdc20 00000000 3fff1a70 40211a11  
3fff19c0:  3fffdc20 00000000 3fff1a70 4020ae38  
3fff19d0:  00000000 00000000 00000000 00000000  
3fff19e0:  00000000 00000000 00000000 00000000  
3fff19f0:  00000000 00000000 00000000 00000000  
3fff1a00:  00000000 00000000 3fff09f0 4020face  
3fff1a10:  00000000 00000000 3fff0a00 40100114  
<<<stack<<<
```

Back to the object file, the address shows:
```
Disassembly of section .text:

40100000 <_stext>:
40100000:	100000        	and	a0, a0, a0
40100003:	ff2140        	excw

40100004 <call_user_start>:
40100004:	ffff21        	l32r	a2, 40100000 <_stext>
40100007:	13e720        	wsr.vecbase	a2
4010000a:	01bc05        	call0	40101bcc <call_user_start_local>
4010000d:	000000        	ill
```

To be honest, I am not the expert in reading stack dumps :) So I am a little bit uncertain whether I'm searching in the right direction or if the problem is elsewhere?

tbnobody, 2015-12-25:
I repeated it several times. Most of the time I get:
```
Fatal exception (0):
epc1=0x4020faa0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
which is:
```
4020faa0 <esp_schedule>:
```

It also happend in the function 
```
static spiffs_cache_page *spiffs_cache_page_get(spiffs *fs, spiffs_page_ix pix) {
```
and in hostap_input on multiple locations: 4022cf7f, 4022cfbf
```
4022cf78 <hostap_input>:
4022cf78:	04bd      	mov.n	a11, a4
4022cf7a:	b0c112        	addi	a1, a1, -80
4022cf7d:	b1f9      	s32i.n	a15, a1, 44
4022cf7f:	91d9      	s32i.n	a13, a1, 36
4022cf81:	7109      	s32i.n	a0, a1, 28
4022cf83:	a1e9      	s32i.n	a14, a1, 40
4022cf85:	81c9      	s32i.n	a12, a1, 32
4022cf87:	02ed      	mov.n	a14, a2
4022cf89:	03cd      	mov.n	a12, a3
4022cf8b:	4c08      	l32i.n	a0, a12, 16
4022cf8d:	0228      	l32i.n	a2, a2, 0
4022cf8f:	0048      	l32i.n	a4, a0, 0
4022cf91:	053d      	mov.n	a3, a5
4022cf93:	145e40        	extui	a5, a4, 14, 2
4022cf96:	45cc      	bnez.n	a5, 4022cf9e <hostap_input+0x26>
4022cf98:	b50040        	extui	a0, a4, 16, 12
4022cf9b:	000106        	j	4022cfa3 <hostap_input+0x2b>
4022cf9e:	1008      	l32i.n	a0, a0, 4
4022cfa0:	410800        	srli	a0, a0, 8
4022cfa3:	f40000        	extui	a0, a0, 0, 16
4022cfa6:	3ab0b6        	bltui	a0, 16, 4022cfe4 <hostap_input+0x6c>
4022cfa9:	1cd8      	l32i.n	a13, a12, 4
4022cfab:	1dd8      	l32i.n	a13, a13, 4
4022cfad:	000d42        	l8ui	a4, a13, 0
4022cfb0:	ffa0f2        	movi	a15, 255
4022cfb3:	145040        	extui	a5, a4, 0, 2
4022cfb6:	048556        	bnez	a5, 4022d002 <hostap_input+0x8a>
4022cfb9:	791c      	movi.n	a9, 23
4022cfbb:	41b9      	s32i.n	a11, a1, 16
4022cfbd:	3139      	s32i.n	a3, a1, 12
4022cfbf:	2129      	s32i.n	a2, a1, 8
```

Links2004, 2015-12-25:
the dumps look like flash access problems, 
to reasons: 
- to less power, what do you use as 3V3 score?
- flash is locked be course of write, erase or read action and needed code is not cached
- your flash is damaged

tbnobody, 2015-12-25:
D'oh... that's embarrassing. I use the USB port with a voltage regulator now for many weeks at it worked like a charm. Just to make sure I didn't miss anything I now power it with my DC power supply. And what can I say? No exceptions / stack dumps any more. 

Thank you very much for your help :)

closed 2015-12-25 by tbnobody

generalgit, 2018-09-23:
> D'oh... that's embarrassing. I use the USB port with a voltage regulator now for many weeks at it worked like a charm. Just to make sure I didn't miss anything I now power it with my DC power supply. And what can I say? No exceptions / stack dumps any more.
> 
> Thank you very much for your help :)

Has this issue been resolved? Was it just the power related problem?