esp8266/Arduino #1300: Automation of my blinds

darkmarc86, 2015-12-24:
Hi guys, I'm looking to automate the blinds in my bedroom for fun. I've got a stepper motor which attaches to the chain of my blinds to roll them up or down. I was hoping to use the ESP8266 to do this. Initially I was using an Arduino Mega but since people have worked out how to flash code onto the ESP, the Arduino is redundant for me. I have had some initial success with the ESP, managing to flash some code onto it and get the wifi LED example working. What I would really like to do is have an app on my android phone with the buttons "close blind" and "open blind" and perhaps an indicator showing the current status of the blind (whether they are open or closed). I would like the messages to go though an internet server so that I can use the app from anywhere. It looks like the website 'thinkgspeak' may do what I want and I've also seen some people using 'pushbullet'. However, I have not been able to find any example code for either that works easily using the Arduino IDE. So firstly, am I going about this the correct way with thingspeak or pushbullet (ie. will they do what I want?) and secondly, can anyone direct me to a resource with some example code for Arduino IDE that communicates with one of these websites? Thanks in advance!

penfold42, 2015-12-24:
I did something similar - a garage door opener.

I wrote a mostly generic gpio web interface :
Http://192.168.1.44/gpio/2/set/0

Personally I don't trust any external service to talk to my home automation so I use a con connect from my phone to home.

Out of curiosity, what hardware are you using to drive the stepper ?

drmpf, 2015-12-24:
My Android pfodDesigner app will let you design your own Android display and generate the ESP code for you to show those buttons on my pfodApp,   See www.pfod.com.au for numerous example projects.
The Garage Door with position sensor (http://www.forward.com.au/pfod/GarageDoorRemote/GarageDoorRemoteWithPosition/index.html) is an example the shows the position on you mobile as well.  (Needs a bit of hand coding).
Happy to help out if you need it. You can contact me at pfod.com.au via support

darkmarc86, 2015-12-25:
penfold42: whats a con connect? The circuit that I'm using to drive my motor is a DRV8825 (current control + chopper circuit) and the motor itself is a NEMA17 2.3A stepper. I've tested them together and it seems to work quite well.

drmpf: Your project and android app look amazing! I'm just a bit worried that because you've made it so noob friendly that it will take the fun out of the project for me. I was hoping to do the coding myself and not have it auto-generated. Though both of you make good points about having control over the security rather than using a 3rd party web server. Matthew, does your app go through the internet or just home wifi?

drmpf, 2015-12-25:
The pfodDesigner only does the basics. Lots more for you to program,  For example the slider showing the position needs to be added afterwards.  
I connect and control the garage door via the internet (see http://www.forward.com.au/pfod/HomeAutomation/ConnectingIoT/index.html for the details)
pfodApp has a special feature to allow connection via the internet address, even when at home.  Normally your router would block such connections.
pfodApp also has a very light weight 128bit security that prevents others from accessing or controlling the device by adding a unique secure hash to each message.
See http://www.forward.com.au/pfod/secureChallengeResponse/index.html for the details.
The GarageDoorRemoteWithPosition is a complete project that uses all of these features.

penfold42, 2015-12-25:
Damn iPhone autocorrect - con was supposed to be VPN.

tzapu, 2015-12-28:
@darkmarc86 i ve done a 2 cahnnel version of what you want with an esp8266 module, relay board and blynk app.
works a treat but never got around to write an article or post the code, if it's of interest to you, i ll put the code up somewhere for you to play with.

looks like this

[image]

[image]

closed 2016-02-29 by igrr