esp8266/Arduino #1307: OneWire library, when use  ESP8266 and DS18B20 with parasite power - pin must be cofigured as push-pull

KAlexK, 2015-12-26:


vlast3k, 2015-12-28:
I tried both - this approach, and invoking the constructor with the internal pullup - in both cases i get 85c.. how have you connected the DS18B20 ? I connect its vcc+gnd to gnd and the data pin to GPIO 14.
There is also another closed issue on this topic, but even also with a resitor it is giving the same results (though i would expect that no resistor is needed for parasitic power)

KAlexK, 2015-12-29:
Yes. Vdd and GND pin of DS18B20 to GND. Data pin (DQ) to GPIO5 and pull-up by 3.3k resistor to power (3.3V). Work fine. I got 85C only if I did not connect Vdd pin of DS18B20 to GND. Try to connect Vdd pin to power and do not use parasite power mode.

igrr, 2016-02-29:
Since we have removed OneWire from this repository in favour of the official version, please open the issue at https://github.com/PaulStoffregen/OneWire.

closed 2016-02-29 by igrr