esp8266/Arduino #1316: WiFi debug not correct

onkelfunny, 2015-12-28:
the wifi-debug show a wrong ip address:
```
IPAddress apIP(192, 168, 1, 1);
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
WiFi.softAP("SSID");
```
Debug-Ouput:
```
mode : softAP(1a:fe:34:aa:4f:33)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bin 100
```
i can connect and have a ip address in the network 192.168.1/24

Links2004, 2015-12-30:
shut be fixed in #1323, there where a mode problem.

Links2004, 2015-12-30:
can you retest it with the latest git?

onkelfunny, 2015-12-30:
the wrong params are still present

Links2004, 2015-12-31:
what is reported by:
```cpp
  IPAddress myIP = WiFi.softAPIP();
    Serail.print("AP IP address: ");
    Serail.println(myIP);
```
the ```dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)```
message is from the SDK so we can not change it.

onkelfunny, 2015-12-31:
WiFi.softAPIP() shows the correct ip

Links2004, 2015-12-31:
ok will close the issues, since we can not fix it.

closed 2015-12-31 by Links2004