esp8266/Arduino #1321: getting data from a POST request

sukhmeet032795, 2015-12-29:
i have a url /wifiSetup. Whenever a user sends post request to the esp, some json is sent along with 
the url. I want to get the json content in my handler

Code:

ESP8266WebServer server1(80);
server1.on("/wifiSetup", wifiSetup);

void wifiSetup(){

  WiFiClient client = server.available();
  while(client.available()) {
    Serial.print(client.read()); // print client data
  }
}

This return me POST /wifiSetup HTTP/1.1
I want the json content sent along with the request. How can i get the json?

sukhmeet032795, 2015-12-29:
For those who are not able to figure it out:

Code:

#include <ESP8266WebServer.h>
ESP8266WebServer server(80);  //creating the server at port 80

void setup() {
  Serial.begin(115200);
  server.on("/yourURL", function); 

//configuring the server wherein you configure your url and function name, you can also mention your method(GET,POST..) if u want i.e server.on("/yourURL", method,function); 

  server.begin();  
}

void function() {
  //do something 
  String varname = server.arg("abc"); //this lets you access the value using the key as you have set in your json for ex:

  
}

void loop() {
  server.handleClient();    //this is required for handling the incoming requests
}

Now find your wifi local ip and make a request at the particular ip and

closed 2015-12-29 by sukhmeet032795

themakerbro, 2016-10-23:
Hey @sukhmeet032795, I'm having problems accessing my JSON data using 

`server.arg("abc");`

It just shows as empty, any chance you can post a simple example?

bkomac, 2016-12-16:
Try
```
StaticJsonBuffer<200> jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(server.arg("plain"));
```

JBarna, 2017-01-18:
For anyone who may be interested, the ESP8266WebServer library does **not** support parsing JSON. You can see [in the source code here](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/Parsing.cpp#L161) that it supports data with MIME types `application/x-www-form-urlencoded` and `multipart/...`. 

So if you want to continue using JSON, you will need to use a jsonBuffer as @bkomac has noted and you can access the JSON string through server.arg("plain");

Otherwise, you can send your data in the supported MIME types. 

Example using `application/x-www-form-urlencoded`
On your webpage you'd send an AJAX request as so
```
var xhr = new XMLHttpRequest();
xhr.open('post', '/postUrl');
xhr.setRequestHeader('Content-type', 'application/x-www-form-urlencoded');
xhr.send('key1=value1&key2=value2');
```

Then on the server you'd access the data as so
```
String val1 = server.arg("key1");
String val2 = server.arg("key2");
```

vladbalanescu, 2017-10-25:
Anyone experiencing this?

[image]

devyte, 2017-10-25:
Try this:
String val1 = server.arg("key1");
String val2 = server.arg("key2");
=>
String val1(server.arg("key1"));
String val2(server.arg("key2"));

From what I've seen, build issues like this usually come from corrupted/bad installation, but in this case that's just a guess. I don't see that error when building my code.

vladbalanescu, 2017-10-25:
Nope, still the same result

devyte, 2017-10-25:
Oh oops. It says server, shouldn't it be webServer?

vladbalanescu, 2017-10-25:
Yes, that is true. Thanks a lot for that spot. My huge mistake there, it was `webServer` in my case

rammanoj, 2018-11-16:
I am sending a request int th following way (using curl)

      curl --request GET --data '{"asd": "asdasd"}' http://192.168.43.141/test

But I am not sure why I can't find any data in args, when I just print `Serial.println(server.args())` it prints empty, and I also tried with `Serial.println(server.arg("plain"));` also prints nothing. I am not sure how the JSON data I sent is getting handled ??

pcbtborges, 2019-10-07:
Hi, what libraries are required to use server.arg?
All I have is   WiFiServer server(80); and it says:
'class WiFiServer' has no member named 'arg'

I am getting all the data I need (SEE BOLD TEXT) when my ESP32webServer gets JSON from a web browser but, so far, I am unable to get to the JSON data on the ESP32 side.

My Problem is:
Get to the JSON data arriving from the browser.

What is displayed at the ESP32 console is:

`New Client.
GET /?**wifiSsid=Skynet&wifiPass=sssssss&delayPrimFoto=sdfgsdg&delayEntreFotos=sfdgsdf&timeGMT=-3&smtpServer=hhhhhhh&smtpUser=jjjjjjj&smtpPass=kkkkkkk&emailto=sfdgsd&emailTitle=gsdfgsdf&emailMessage=&ftpServer=&ftpUser=&ftpPass=** HTTP/1.1
Host: 192.168.4.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
DNT: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.90 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3
Referer: http://192.168.4.1/?wifiSsid=Skynet&wifiPass=babacabaca&delayPrimFoto=1&delayEntreFotos=3&timeGMT=-3&sendtoemail=checked&smtpServer=&smtpUser=&smtpPass=&emailto=&emailTitle=&emailMessage=&ftpServer=&ftpUser=&ftpPass=
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9,pt;q=0.8,sv;q=0.7,es;q=0.6

Client disconnected.
`
Assistance welcome
Paulo

devyte, 2019-10-07:
@pcbtborges wrong repo, this is for the esp8266, not the esp32.

pcbtborges, 2019-10-07:
Ok, I see, but I was convinced the problem applies for both.
Thanks

devyte, 2019-10-07:
That's possible, but the code base, repo, dev teams, and policies are very different. If true for our esp8266, please open a new issue and fill out the required info.

earlephilhower, 2019-10-07:
` WiFiServer` is NOT a web HTTP class.  It's a basic TCP socket class.  SEnds and received raw bytes.

ESP8266WebServer or ESP8266HttpClient are the HTTP protocol classes.

pcbtborges, 2019-10-08:
Hi, thanks, I posted this question on the Espressif forum as well.
"WiFiServer is NOT a web HTTP class."
I realized that after some research.
But I reached the following command:
clientPayload = header.substring(header.indexOf("GET /?")+6,header.indexOf("HTTP/1.1"));
And with that I successfully extracted the long string that has everything I need.
Now, it is going to be a lot of work to split all that out to the variable I need.
I would prefer to use a more professional approach like JSON but I still lack the understanding to switch from WiFiServer to a real web server library.
How would I extract the JSON from the payload I get from the customer browser?

Sorry for posting on a "closed" topic that in fact are for ESP8266 but that was the one I found that was the closest match for my problem. It would be a waste to live this knowledge unused.

Thanks everyone.

SusyVenta, 2023-11-03:
> StaticJsonBuffer<200> jsonBuffer;
> JsonObject& root = jsonBuffer.parseObject(server.arg("plain"));

To expand on this, the following works for me:

```
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
...

server.on("/changeAlarmStatus", HTTP_POST, changeAlarmStatus); 

...
void changeAlarmStatus(){
  String data = server.arg("plain");
  StaticJsonDocument<400> jDoc;
  DeserializationError err = deserializeJson(jDoc, data);
  JsonObject jObject = jDoc.as<JsonObject>();
  String status = jObject["status"];
  ...
}
```

POST request:
```
curl -X POST --header "Content-Type: application/json" --data '{status: "on"}' http://<IP>/changeAlarmStatus
```