esp8266/Arduino #1327: WiFi.SoftAP(ssid, password) still not working in v2.0.0

EnochHwang, 2015-12-30:
I read in other posts that v2.0.0 has fixed the problem with softAP not changing the ssid and/or password. It was working for me last week, but now it is no longer working. Whatever ssid I use in the WiFi.softAP command, the WiFi I get is always the original ESP_DAFA99. I'm using Arduino IDE v1.6.7. How should I debug this?

tzapu, 2015-12-30:
i can cofirm the same behaviour @EnochHwang , you need to set a valid password for the configuration to be updated.

Links2004, 2015-12-30:
in #1323 i add a check for this and softAP will return false if password is to short.
https://github.com/esp8266/Arduino/pull/1323/files#diff-49f133721a21472dcdbc6beb4394f329R96

closed 2015-12-30 by EnochHwang