esp8266/Arduino #1328: ets Jan  8 2013,rst cause:4, boot mode:(3,7)

lhcavalcantis, 2015-12-30:
I tried to resolve the reseting problem with the previous issues here. But I already correct all the pins ( GPIO0 pulled up, GPIO15 pulled down, rst and ch_pc also pulled up) and the problem continues. So I checked the supply, but I'm using a lm1117 and the voltage provided its also correct.

Some issue before talked about disabling the watchdog ( ESP.wdtDisable();), I  tested and the problem persists.

After the problem line appears:

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Anyone knows what can I do to fix it?

me-no-dev, 2015-12-30:
So how do you get to trigger that problem? Please post some code that triggers it. The issue as explained could not be understood.

lhcavalcantis, 2015-12-30:
Sorry.
I'm trying to use the WebClient example to send the temperature from my work space. So I just changed the value sent and the frequency of upload data.

The code is:
´´´
#include <ESP8266WiFi.h>

const char* ssid     = "------------------------";
const char* password = "-------------------";

const char* host = "data.sparkfun.com";
const char* streamId   = "G2NoXQalpKHjJ0ZvM7GL";
const char* privateKey = "----------------------------------";

void setup() {
  Serial.begin(115200);
  delay(10);
  // We start by connecting to a WiFi network
  //ESP.wdtDisable();
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int temp = 0;

void loop() {
  temp = analogRead(A0); // reading sensor
  temp = temp * 0.09765625; // costance (1/1024*100)
   
  
//  ++value;

  Serial.print("connecting to ");
  Serial.println(host);
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  
  
  // We now create a URI for the request
  String url = "/input/";
  url += streamId;
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += temp;
  
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(10);
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    // Serial.print(line); // Not using this message yet 
  }
  Serial.println("time");
  delay(20000); // Just send every 20 seconds
  Serial.println("closing connection");
  Serial.println();
}
´´´

demandajmenso, 2016-01-01:
Check this out maybe we have the same issue.
https://github.com/esp8266/Arduino/issues/1342

demandajmenso, 2016-01-01:
Also I have added a large capacitor (470uF) across the Vcc and Gnd rails

TwatBurglar, 2016-02-05:
OP, did you find a solution for this?     wdt_disable() DOES NOT WORK, but it DOES seem to feed the watchdog.   8 seconds after wdt_disable(), watchdog will always time out.

closed 2016-02-13 by lhcavalcantis

pratikmokashi, 2016-04-22:
having same problem 
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

stsrc, 2016-04-27:
Be sure that your power supply is strong and stable enough. To get it working with stm32f0discovery board I had to get:

- anexternal voltage source (12V),
- voltage stabilizer to 5V (with radiator!),
- voltage stabilizer to 3V3, 
- 2200uF(!) electrolytic capacitor on 3V3, as close as possible to esp8266 power suply pin. 
This capacitor may be way to big, but I am only prototyping and want things to work.
- 100nF ceramic capacitor as close as possible to esp8266 power suply pin.

and it finally works correct. 

- Note about stabilizers - also connect to them some capacitors as their datasheets say.
- Note about big capacitor - esp8266 may work with smaller capacitor, just test it. Electrolytic capacitor should have low impedance.

Captayne, 2016-05-28:
I have the same problem, and can't get rid of it. I am close to throw all ESP8266 away and go for another IoT system on chip.
I can't get this running stable: https://github.com/cnlohr/esp8266ws2812i2s
I use a PC power supply with 3.3V capable to drive 25A. Also added some ceramic capacitors to VCC and 3.3V directly on the pins of the Nodemcu accumulating now in up to ~3700µF.
From time to time (~~ every 20 minutes ) the chip resets with the following serial output:
_ets Jan  8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld_

I feel soo helpless now, because everything else works fine. Meanwhile i Have also added some other hadware and software featrues and made a new website to control pumps for a bar mixing app. 
Is there anything else to debug and look deeper? 
Maybe there really is a bug in the Fastled Library or any other lib used in the example?
I use Arduino 1.6.5.

Thanks for any hints.
Andreas

krzychb, 2016-05-28:
@Captayne

Do you have issue with application from https://github.com/cnlohr/esp8266ws2812i2s or "Fastled Library" from https://github.com/FastLED/FastLED?

Captayne, 2016-05-28:
Actually, I don't know. I just experience the crashes. If someone can give me a hint, to track down the reboots, I would be more than lucky. 

Mit besten Grüssen/with my best regards,
Dr.-Ing. Andreas Keibel
SYSTRAGON 

Authored and sent using Lumia 1020

https://github.com/esp8266/Arduino/issues/1328#issuecomment-222321264

Captayne, 2016-05-29:
I now have changed the power supply to another PC power supply, with lots of amps on 3.3V. But reboots still occur occassionally. Always this shows up in the usb serial monitor:
_ets Jan 8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld_
What does room 16 mean?
The websocket error event results in #1006. (quite clear.)

Of cause I can reconnect easily from JScript, but this in not a honest solution. 

Any idea? Can the code be buggy, resulting in such an error? If the code is buggy, then the beaviour should be little more deterministic and not with crashes from time to time. Website and ESP do communicate in 10Hz frequency. all works fine without any overlap until the ESP crashes. 
I have swapped the NodeMCU Devboard with 5 others without any change. I put it from my PCB design back into the protoyping board with no change in behaviour. 
Has somebody similiar problems?
Is there a possiblity to ad more detailed output on what the system crashes? Memory leak/ Nullpointer assignement/ memory read error / invalid instruction/  page error (I think rather not, as there is no paging).... ?
uh!?
Any hints?
Thank you soo much.

krzychb, 2016-05-29:
@Captayne,

1. If not done already, you may check recommendations [here](http://internetofhomethings.com/homethings/?p=396) 

2. To verify if watchdog timer resets may be caused by your application, I would load one of example sketches that is using WiFi, e.g. [WiFiWebServer.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiWebServer/WiFiWebServer.ino) and leave it running for couple of hours.

Krzysztof

akashdeep3194, 2016-08-01:
I was facing this problem and it was fixed by changing Tools>Upload Speed to 115200.
And Flash Size to 4M(3M SPIFFS).

hadipic, 2016-09-03:
eset ets Jan 8 2013,rst cause:4, boot  
eset ets Jan 8 2013,rst cause:2, boot  
 
add this code slove  problem all reset
ets_intr_lock(15); // all intruptt off
 ets_wdt_disable();

jumpjack, 2016-10-05:
I have same issue.
How can I specifically add ESP-07 to boards available on Arduino IDE?

jumpjack, 2016-10-05:
I'm using this setup: 
[image]
http://ediy.com.my/blog/item/133-upload-sketch-to-the-esp8266-esp-07-esp-12-using-arduino-ide

lagunacomputer, 2016-11-04:
was able to fix this by changing the Digital Pin Assignment of my onewire device to D4, and my DHT sensor pin assignment to D6.  it did not like the previous pin assignments.

RubenVot, 2016-11-13:
I had same problem and same message. I was trying to set the D8 to INPUT. 
Changed to 0, everything ok now :)

cesarcardinali, 2017-01-09:
Guyyyyyyys! Problem solved!
Connect an external power source to Arduino and it should work just fine!
I plugged an 9v 1A and at same moment it came to work!

isaiasmoioso, 2017-01-20:
@cesarcardinali I"m not sure i follow. Did you mean you plugged in an external power source to your esp8266 to get it to work?

RubenVot, 2017-01-20:
Sometimes cheap power sources can make that arduino behaves in an erratic mode.

cesarcardinali, 2017-01-20:
@isaiasmoioso Exactly! Buy an external power source capable of 1000mA and it should work just fine.
I am using it connected to my Arduino and an external power source. Problem solved.

macmeck, 2017-02-11:
I was having the same issue while my power supply was just fine. Solved it by using a different USB to Serial adapter. FTDI adapters brought up this issue (while also making trouble during programming from time to time) whereas I had no issues using a CP2102 adapter. You might want to give this a try...

gojimmypi, 2017-02-12:
fwiw - I found this thread while searching for the cause of my continual ESP8266 reboots - the "rst cause:2, boot mode:(3,7)" messages out the serial port at 74880 baud - followed by repeated diagnostics, only to repeat.

My issue, is that I believe my programming method was forcing in corrupted code into the ESP8266.

The solution for me? Hold down the "Flash button" while programming. Previously I had been using VisualMicro to program the NodeMCU via the UART. (the UART handles the flash control automagically) I knew I had a good device, as I could successfully program it via the UART. However I wanted to use a JTAG debugger such as the Segger or Olimex with VisualGDB - but could never get it to work with the NodeMCU. (I spent a ridiculous amount of time before I found this solution).

While googling, I came across this: [https://nodemcu.readthedocs.io/en/master/en/flash/#putting-device-into-flash-mode](url) I had never needed to press the flash button, as the USB connection would do that for me automatically. But with JTAG I no longer had this luxury! Note that the flash button needs to be held down *before* power is applied to the ESP8266.

Hopefully this will help someone else & save you a few hours...

lagunacomputer, 2017-02-12:
lol 74880, what was that like the insta-obsolete V.32fast? lol anyway i gotta tell you, i have bought about 10 nodemcu's from ebay, and it seems half were 'legit' and the other half we cheap knockoffs that did not do so well in the 'serial communication' and overall 'works great' departments.   Some would work fine with the arduino 1.6x IDE and a standard stupid usb cable, others would just fight me to load a firmware.  I think the cheapie chinese knock offs are a problem.  The only way i could visually disingush between the two, the 'print' on the big ESP8266 chip on the fake NodeMCU's from ebay was visually rough and 'falling off'.  Check yours, do you have the same cheap paint on your ESP8266 chip on your NodeMCU?

lagunacomputer, 2017-02-12:
and YES YOU MUST SUPPLY EXTERNAL POWER if you want to run anything other than a LED off of the pins of a NodeMCU

Captayne, 2017-03-16:
Guys, after almost a year of doing nothing on my project, which came from little demotivation, and speculating about errorfixing in the IDE and libraries by others, ... 
... NOW really time seems to have worked for me. 
I upgraded from Arduino 1.6.5 to 1.8.1 and compilied and flashed again.  Everything in the Dev-Environment and libraries were upgraded. 
AND ....   

Tadaaa! No reboots occour anymore! 

WebSocket connection remains stable, and the ESP runns like hell for days.
Yiiippiii!

Captayne, 2017-03-16:
It crashed again,  with "ets Jan  8 2013,rst cause:4, boot mode:(3,7)" .... not after days but after 5 hours. .... but after another 5h it still runs. .... The error is not resolved. what can it  be?

aminbehravan, 2017-03-24:
i tested, just change flash size, it will be fixed. it was really complicated

Captayne, 2017-03-25:
Change flash size? Please give little more advice. Thank you. 
Captayne.

aminbehravan, 2017-03-28:
if you are using the esp-12 as i used, just go to "tools" and then "flash size" and choose one of them. 
"4M (1M spiffs) worked for me.

bobybobybob, 2017-04-05:
Hello ESP8266 Friends,

i have try erase firmware and upload other one. no work well. 

i have 3 x 12E.   Two are stable and ok and one make much trouble. Perhaps i have too hot for too long soldered on adapter board. 

My script is very simple... only show number +1 in Serial and the trouble one hang fast. 

I belive now it is Hardware defective

Captayne, 2017-04-13:
The adjustment to "4M (1M spiffs)" worked for me too!  Thank you AminBehravan!  
I was so badly debugging my stuff, ... but sometimes the solution is soo simple but on a far planet at the same time.
Now it works and my project can go on. My trust in ESP is back.
(btw. can't the spiffs whatever seettings can't be detected and set automatically?)

Best Andreas

gojimmypi, 2017-04-24:
fwiw - while working on my "build esp8266" script:

[https://github.com/gojimmypi/build_esp8266](url)

I discovered the most recent version of esptool.py here:

[https://github.com/themadinventor/esptool/](url)

no longer requires explicit flash size, as it auto-detects it! :)

pipoblak, 2017-05-29:
Im getting this error after a while using FASTLEDS in my nodemcu. Anyone knows how to resolve this?

the code:
https://gist.github.com/pipoblak/344c8b93625dff744fd296a73a78da38
the stack trace:
```

Connecting to QG BOLADAO
......
WiFi connected
IP address: 
192.168.1.76
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
chg_B1:-40
chg_B2:0
mac 981

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
þ
```

pipoblak, 2017-05-29:
I put the yield() in frequently used functions and don't work too :(

pipoblak, 2017-05-30:
I tryed a more simple code with FastLeds and NodeMCU and dont work too.

Code: https://gist.github.com/pipoblak/9e164e86242728e14f72500d2de9eeb4

StackTrace:
```
dev 1143

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
á

```

scotthibbs, 2017-06-26:
wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

I solved this error loop because I had 
#define DOOR_PIN 6
instead of 
#define DOOR_PIN 4

So check which pins are assigned. 
Hope this helps,
Scott

NarreKiwi, 2017-07-15:
This problem entirely goes away, after completely erasing the external SPI FLASH chip. Most likely it got corrupted somehow along the way from the previous FLASH upload to the chip, since it triggers the WDT reset and keeps rebooting all the time, after you pressed the reset button on the ESP12-E which I've got.
You need to use a short Arduino IDE sketch to do it - see https://github.com/kentaylor/EraseEsp8266Flash
Then I used the NodeMCU Flasher to upload a normal program *.bin file to it, to run a normal serial comms program from it which I checked the coms through the Arduino IDE serial monitor.

shitalshivani, 2017-08-12:
Hi everyone!!
i have ESP12-E,i am using this for IOT application.I have done with my code.Now i am testing this code on esp8266 12E node MCU board.and i am getting random error of exception 0 and
 ets Jan  8 2013,rst cause:2, boot mode:(3,7) wdt reset after 24 hours and sometimes it occurs after few hours.
I am not getting whats wrong with my code.please suggest any solution.thanks in advance.

NarreKiwi, 2017-08-13:
Most likely fault caused by programming corruption after you FLASHed it. Try reprogramming it with an original factory  .bin file.
-- 
Sent from my Android phone with mail.com Mail. Please excuse my brevity.

raufjay, 2017-11-10:
I had the same issue which is resolved now.

dt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld

Just resolved it with a strange solution.

I was using example of basicOTA with a minor addition of blink. In the loop I had a condition to check pin D8, if LOW go to the OTA handler so it will wait to receive sketch OverTheAir. If D8 is HIGH then continue blinking LED.

After uploading the code via com port, the esp8266 or nodemcu should blink the led unless i pull down the D8. But I kept getting the above error.

I tried using D7 and D3 as well, with the same result.

I was using "#define BUTTON  8" statement.

But when I used "#define BUTTON D8" instead, all issues gone, and esp8266 started working normally.  

Strangely, the pin D2 which is used for the built-in LED is still flashing ON and OFF without using letter D. I can still use digitalWrite(2,HIGH) or LOW. That means there is some inherent problem with the IDE porting code which behaves different for the INPUT and OUTPUT modes.

Can some one explain it to me?

Hope it may help someone.

Thanks,

Rauf

Tvde1, 2018-05-14:
I'm getting this issue without having touched the board before.

I bought an ESP8266 and connected to it with the serial monitor and got this. I've tried flashing the basic Blink demo but that isn't fixing anything. I've played around with all the settings.

Any help?

lagunacomputer, 2018-05-14:
Try flashing a blank sketch

man6663, 2018-06-09:
resolve it by change to external power supply....usb port not enough amp

kopolindo, 2018-08-13:
@akashdeep3194 GOD BLESS YOU. I owe you a beer

daekonn, 2018-09-11:
I solved...
by  inserting
=====================
   ESP.wdtDisable();
=====================
inside of the problem making function ,   not in the setup().  ^^

devyte, 2018-09-11:
> @daekonn which function?

ivm164, 2018-10-18:
I've got resets when trying to use quite large local array. In my case it was bool a[150][150]. IDE doesn't produce any warning during compilation in this case. When I made that array global I've got the warning on possible instability due to low memory but in the end there were no real problems and at least I know now what I should optimize to make things better.

dehuy69, 2018-11-07:
Try:
In Arduino IDE Tools >> Erase Flash >> All flash content
recompile and upload

frustrat01, 2018-12-08:
Having some issues with the ESP8266-01 by uploading the sensor data to the localhost.

The below shows the issues when I open the Serial Monitor in the Arduino
` ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset
`
By the way, I have tried using the WiFiClient example to upload the dummy data to the localhost and it works. But it cannot work with uploading the sensor to the localhost. 
Please help me to get rid of this issue ASAP

frustrat01, 2018-12-08:
Having an issue with the ESP8266-01 by uploading the sensor data to the localhost.
The below shows the issues when I open the Serial Monitor in the Arduino

` ets Jan 8 2013,rst cause:4, boot mode:(1,0)

wdt reset
`
By the way, I have tried using the WiFiClient example to upload the dummy data to the localhost and it works. But it cannot work with uploading the sensor to the localhost.
Anyone has the solution please help me to get rid of this issue ASAP. Thanks

frustrat01, 2018-12-08:
@lagunacomputerI don't understand what you mean. Can you give the example for this because I still new to this Arduino and ESP8266-01.

the below text file is my Arduino source code for this
[ESP8266_upload.txt](https://github.com/esp8266/Arduino/files/2659928/ESP8266_upload.txt)

lagunacomputer, 2018-12-08:
Pretty sure esp8266 only has 2 gpio pins, looks like you are trying to use more than actually available, I bet the code works on a Uno or a mega because they support more pins

“#define in 7
#define out 8define #trigPin 10 #define echoPin 13 #define relay 2 #define led 6 //red color #define led2 A0 //yellow color #define led3 9”

frustrat01, 2018-12-08:
I use 1 GPIO 0 in esp8266, but I didn't declare it in the source code

devimeruva, 2019-01-22:
17:34:15.107 -> 
17:34:15.107 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,6)
17:34:15.107 -> 
17:34:15.107 -> wdt reset

this is being displayed on my COM port instead of showing IP Address. How can i resolve this issue?

lagunacomputer, 2019-01-22:
Change to a different pin, use different PIN numbers in your code

lrmoreno007, 2019-01-22:
It's documented
https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#watchdog

KiLLAAA, 2019-01-24:
@devimeruva  post code?

devimeruva, 2019-01-24:
#include <ESP8266WiFi.h>
#include <LiquidCrystal.h>
#include <EmonLib.h>
EnergyMonitor emon1;             // Create an instance
LiquidCrystal lcd(12,11,5,4,3,2);

const char* ssid = "Zunik Energies Pvt Ltd";
const char* password = "zunik1234";
WiFiServer server(80);

void setup() {
  Serial.begin(9600);
  Serial.setDebugOutput(true);
  lcd.begin(20,4);
  emon1.voltage(2,2313.26,1.7);
  emon1.current(1,58.1);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.write(13);
  server.begin();
  Serial.println(WiFi.localIP());
}

void loop() {
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  Serial.println("new client");
  while(!client.available())
  {
    delay(1);
  }
  emon1.calcVI(20,2000);
  emon1.serialprint();
  unsigned int long timemillis = millis();
  unsigned int long time = millis();
  float realPower = emon1.realPower;
  float apparentPower = emon1.apparentPower;
  float supplyVoltage = emon1.Vrms;
  float Irms = emon1.Irms;
  float powerFactor = (realPower/ apparentPower);
  lcd.setCursor(4,0);
  lcd.print ("EnergyMonitor");
  lcd.setCursor(0,1);
  lcd.print("V = ");
  lcd.print (supplyVoltage);
  lcd.setCursor(9,1);
  lcd.print("I = ");
  lcd.print(Irms);
  lcd.setCursor(0,2);
  lcd.print("Powerfactor = ");
  lcd.print(powerFactor);
  lcd.setCursor(0,3);
  lcd.print("Units(kWh) = ");
  lcd.print(abs(realPower*time)/abs(1000*3600));
  String web = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
  web += "<html>\r\n";
  web += "<body>\r\n";
  web += "<h1>ENERGY MONITOR</h1>\r\n";
  web += "<p>\r\n";
      web += "V=\r\n";
      web += Serial.print(supplyVoltage);
      web += "I=\r\n";
      web += Serial.print(Irms);
      web += "Powerfactor =\r\n";
      web += Serial.print(powerFactor);
      web += "Units(kWh) = \r\n";
      web += Serial.print((realPower*time)/abs(1000*3600));

   web += "</body>\r\n";
  web += "</html>\r\n";

  client.print(web);
}

[15 more replies]