esp8266/Arduino #1331: WiFi and I2C interfere

Nurgak, 2015-12-30:
I need WiFi and I2C/Wire in the same project, but it seems when the `Wire` library is included *and* initialised the WiFi stops behaving as expected. I've been trying to debug this issue for 3 days and now I'm stuck.

It seems that WiFi and I2C interfere somehow. Let me explain: I open the `SDWebServer` example file, run it vanilla and get the following output:

    Connecting to 4a82dd
    scandone
    state: 0 -> 2 (b0)
    state: 2 -> 3 (0)
    state: 3 -> 5 (10)
    add 0
    aid 3
    cnt 
    
    connected with 4a82dd, channel 11
    dhcp client start...
    chg_B1:-40
    chg_B1:-80
    ip:192.168.0.20,mask:255.255.255.0,gw:192.168.0.1
    Connected! IP address: 192.168.0.20
    MDNS responder started
    You can now connect to http://esp8266sd.local
    HTTP server started
    SD Card initialized.
    pm open,type:2 0```

This is expected, it has always run this way for me. For some reason it loses connection (state `5`), but then it connects to my access point and everything is good.

Now when the `Wire` library is included and `Wire.begin(4, 2)` called inside `setup()` (doesn't matter at which point) I get the following output from the ESP:

    Connecting to 4a82dd
    scandone
    state: 0 -> 2 (b0)
    state: 2 -> 3 (0)
    state: 3 -> 5 (10)
    add 0
    aid 3
    cnt 
    chg_B1:-40
    
    connected with 4a82dd, channel 11
    dhcp client start...
    chg_B1:-80
    Could not connect to4a82dd
    pm open,type:2 0

As you can see it fails to connect, simply because I included and initialised the `Wire` library. This happens pretty much systematically, sometimes it succeeds in connecting, which is really worrisome. I tried different pin combinations for `SDA` and `SCL` signals, but it doesn't seem to improve the situation.

Can anybody try to recreate and confirm this? Open the `SDWebServer` example, include the `Wire` library and initialise it inside `setup()`. [Here's a link to Pastebin](http://pastebin.com/u9QUQu89) with the modified version of `SDWebServer` code.

It could be that my flash chip is busted as well, I should test that somehow...

BuzzBurrowes, 2015-12-31:
I am using the Wifi and the Wire library without a problem. I am using pins 5 & 4 for I2C. I am using the latest git version, not the 'stable' version.

Nurgak, 2016-01-01:
I'm using Arduino 1.6.7 and the latest Git version of ESP library as well.

I'll try with another chip as soon as I'm back home in a few days, this really feels like a hardware issue. I have reprogrammed my ESP a few hundred times now and I've heard it's not the best quality flash, so it just might be that (as unlikely as it sounds).

andreaslink-de, 2016-11-12:
Hej Nurgak, was that problem somehow solved? I found this thread struggeling with a compareable problem. I'm using an ESP8266-12F and have a SHT21 sensor via I²C connected. First I also thought it was related to WiFi-related actions in parallel but then I removed everything and simply let the ESP only do requesting the SHT21 for two values and report results to Serial-IF. I can always read once (either temp or humidity) but get crappy values and when I try to read the second value or a second time the ESP is stuck with executing the last row "**pm open,type:2 0**" which I cannot really interpret:

```
Reading temperature from SHT21 sensor... successful
Reading humidity from SHT21 sensor...
pm open,type:2 0
```

As I can find the same row in your report above it seems to be related to your issue, so I'm verly much looking forward to your solution.

Nurgak, 2016-11-13:
Hi andreaslink-de. The issue went away by itself via an update of the ESP libraries. For reference the working code, using WiFi and I2C is [published here](https://github.com/Nurgak/Internet-of-Things-Power-Meter).

closed 2016-11-13 by Nurgak

riccardo1991, 2016-12-13:
I do not know if this could be useful: I added an RTC (DS1307) that uses I2C to my project and totally randomly I had problems connecting with the WiFi (my ESP is going to deep sleep and then wakes up to reads some sensors).
Without RTC, everything works smoothly, but with it the connection fails so many times, around 80% of my tries.

I solved using `WiFi.mode(WIFI_OFF);` ad the beginning of my setup function, and then `WiFi.begin(SSID, PASSWORD);` when I really need to connect.

thieuanh1995hn, 2018-08-13:
Same issue when using both esp wifi with I2c lcd 16x2

RomanGrekov, 2020-01-31:
Hi. I have the same issue. My setup: i2c Lcd 16x2 with esp32.

gijsneerhof, 2020-03-11:
I too have this issue. Whenever my ESP8266 12F module talks to / receives data from the I2C device, the WiFi disconnects/gives trouble. Could it be an EMC issue?