esp8266/Arduino #1335: esprduino with hx711 not working

EagleShi, 2015-12-31:
HX711 library
https://github.com/bogde/HX711/blob/master/HX711.h

HX hx(12,14);

12,14 not working

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
pm open phy_2,type:2 0 0
cnt

igrr, 2015-12-31:
There is a possibly infinite loop inside read function in this library:
while (!is_ready());

This won't work well on the esp8266.

Links2004, 2015-12-31:
```cpp
	// wait for the chip to become ready
	while (!is_ready()) {
	    delay(0);
	}
```

EagleShi, 2015-12-31:
I added "delay (0)", but still does not work

EagleShi, 2016-01-01:
i found in boot, GOIP 12, and GPIO 13 hight, but I need to hx711 as of the beginning of the GPIO 12 LOW,
Normal mode: 12=1 GPIO, GPIO 13 =1
boot with HX711:Gpio12=0 GPIO, 13=1 
the hx711 Code is :
SCK=12 byte; DOUT=13 //byte;
PinMode (SCK, OUTPUT);
PinMode (DOUT, INPUT);
DigitalWrite (SCK, LOW);

After the hx711 initialed, esp is wdtreset , 
add the ESP.wdtDisable (), but still does not work

my english is poor ,thank you for your help.

igrr, 2016-03-23:
Closing this as duplicate of #1809

closed 2016-03-23 by igrr

emericspiroux, 2017-10-13:
add delay(0) inside loop work for me ! Thanks