esp8266/Arduino #1337: Arduino wifi AVRISP

wifixcort, 2015-12-31:
Hi, I'm having trouble trying to use the Arduino WiFi AVRISP example, I can see than I'm having connection with ESP8266, but the signature is not correct when it try to program. It could be a problem with timing to restart or something.

I could see information that looks interesting

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

ioctl("TIOCMGET"): Inappropriate ioctl for device

Anyone have any idea what might be going

igrr, 2016-02-29:
/cc @kzyapkov

kzyapkov, 2016-02-29:
The `ioctl` error seems to be the norm. `avrdude` tries to treat a TCP socket as if it's a serial port, so some `ioctl()` calls fail. This doesn't seem to affect functionality.

If you're getting all 0x00 or 0xFF for device signature i'ts very likely this is a wiring/power issue. Make sure all required lines are hooked correctly and the reset is configured to trigger with the correct level. 

Otherwise, try decreasing the SPI clock frequency.

wifixcort, 2016-02-29:
Thanks :) 
I'm not sure it's the wiring, because I'm using it on other things, but I going to check again

I'll try and see if I can get it to work