esp8266/Arduino #1339: Make platformio happy by automatically generating version.txt file

sticilface, 2015-12-31:
A version.txt file is required to keep platformio happy.  

When it is placed in the project root the entire esp8266 directory can be symlinked into platformio and it works with no modification.  

I Can just add a version.txt file, but it might be nice to have one generated that matches the project version or git.. if this is possible?   Can just contain a 1 if not..

Links2004, 2015-12-31:
you can add a call like this:
```bash
ver=`git describe --tags --always`
echo esp8266-$ver > version.txt
```
(this is bash not python)

to tools/get.py

sticilface, 2015-12-31:
Would this be a way to do it for everyone?  

ps.  i've not touched python, ever... in fact i tried once and broke everything...

sticilface, 2016-01-02:
Further discoveries.  This file should actually contain the arduino version so in this case 
10606.  Any objections to putting this file in the project root?

igrr, 2016-01-02:
The core is not bound to a particular version of the IDE. It would be a bit
strange to include a version.txt with a value which doesn't match core's
version. Perhaps we can generate this file automatically from platformio?

ivankravets, 2016-01-22:
@igrr where I can get valid version? What is it? https://github.com/esp8266/Arduino/blob/master/platform.txt#L9

igrr, 2016-02-29:
We can generate version.txt inside the released archive automatically, if it helps. Alternatively, you can parse the release archive name (it looks like `esp8266-<version>.zip`).

ivankravets, 2016-02-29:
I see that you have this info in `platform.txt`. We will use it. See https://github.com/platformio/platformio/issues/548

You can close this issue.

igrr, 2016-02-29:
Ah, okay, then we need to make sure the version in platform.txt matches the real version.

ivankravets, 2016-02-29:
@igrr we have interesting issue https://github.com/platformio/platformio/issues/479 where we will propose different vendors to create own development platform compatible with PlatformIO. In this case, this platform will depend on the different pre-built packages and will contain own build scripts.

Ideally, will be good to have some manifest file per package. Package is generic meaning. JS developers mean under it node/npm based package, Python devs mean under it PyPi package and etc.

How about to place this info to the repo?
`package.json`
```json
{
  "name": "framework-arduino-esp8266",
  "description": "ESP8266 core for Arduino",
  "repository": {
    "type": "git",
    "url": "https://github.com/esp8266/Arduino.git"
  },
  "version": "2.1.0",
  "license": "GPL v2.0",
  "authors": [
    {
       "name": "Ivan Grokhotkov",
       "email": "mail@example.com"
    },
    {
       "name": "Other Author",
       "email": "mail@example.com"
    }
  ]
}
```

You correct any fields. This will allow to have independent MANIFEST for the different IDE, build systems and etc.

Comments are welcome.

igrr, 2016-02-29:
Does this file have to be in the repository, or just in the released zip package? I'm not really familiar with PlatformIO internals. Will it really use this git URL provided in manifest, or will it use a released zip package? Released package has a number of differences from the contents of git repository.

ivankravets, 2016-02-29:
I propose to place it directly in the repository. This file will not have any relations to PlatformIO. It can be used by other build system, IDEs and etc.

> Released package has a number of differences from the contents of git repository.

Hm... I didn't know. How do our users use the latest ESP8266 core using `git clone`? Which are difference?

igrr, 2016-02-29:
The difference is in the way the package is installed.
When users are installing latest git version, it is placed into Arduino
hardware folder. When users install the released boards manager package, it
is installed by Arduino into AppData. IDE handles tool paths differently
for these two cases, so we make modifications to platform.txt while
releasing the package to support both installation methods.

Regarding the package json — which version should we place into the
repository? Let's say we have released 2.1.0, should we then modify the
version in package json to something like 2.2.0-git?

sticilface, 2016-02-29:
For what it is worth, I have the repo git cloned and i symlink it into the Arduino Hardware folder.  

ideally for me i would like the same thing with Platformio, as I'm always testing different changes to the IDE, working with different branches, testing things etc.  

I've actually gone back to the IDE, as platformio seemed to need special linkers, and when they change it broke things.. my fault for being on the bleeding edge.  But thats my use case.  

I don't install via package managers, as i rely on git too much for switching branches of everything!

ivankravets, 2016-02-29:
> The difference is in the way the package is installed. ... Arduino IDE

I've got you. That doesn't have relation to PlatformIO because we use raw source files and have own build system. The cloned repo is Ok for us.

>  Let's say we have released 2.1.0, should we then modify the
version in package json to something like 2.2.0-git?

It depends on your workflow. We use https://github.com/nvie/gitflow for @PlatformIO. The releases are sticked in `master` branch and `version` is equal to `x.y.z`, where development process goes within `develop` branch and here `version=x.y.z(.devN|aN|bN|pre|rc)`. For example, https://github.com/platformio/platformio/blob/develop/platformio/__init__.py#L17

If you decide to use SINGLE `master` for `develop` and `releases`, then you can do as you have described above. In the master repo `package.json>version=2.2.0-alpha.1` (according to [semver](http://semver.org)) and when you will build final release then "in the archive" need to replace it to  `package.json>version=2.2.0`

ivankravets, 2016-02-29:
@sticilface 
> I've actually gone back to the IDE, as platformio seemed to need special linkers, and when they change it broke things.. my fault for being on the bleeding edge. But thats my use case.

Please [open an issue for that](https://github.com/platformio/platformio/issues). We will think on it how to use original LD scripts from the repo. It is related to the different build systems. PlatformIO has own build system and the final object files have others location and names

P.S: I hope we will not have significantly difference when update to 2.1.0 release. See https://github.com/platformio/platformio/issues/544 . After this you will have ability to use `clone repo` without modifications to LD scripts.

ivankravets, 2017-08-11:
Resolved in https://github.com/esp8266/Arduino/pull/3321

closed 2017-08-11 by ivankravets