esp8266/Arduino #1342: Spaces in SSID and password resets esp8266

demandajmenso, 2016-01-01:
Can anyone confirm this? After some head scratching i finally figured out why my module resets or doesn't respond in some cases when i try to connect to an AP which has spaces in SSID and/or password.

celsworth, 2016-01-01:
As a point of reference, my AP password has spaces in it (but my SSID doesn't) and I've never seen a reset because of it.

WheresWally, 2016-05-04:
I had the exact same error because I had a trailing space after the SSID,

devyte, 2017-10-19:
I can't reproduce this with latest git. Connecting to my Android hotspot (ssid has a space) works fine.
Closing as can't reproduce.

closed 2017-10-19 by devyte

nalinda6963, 2018-07-10:
@AalianKhan did you find the issue with your sonoff. I am having a similar issue with sonoff basic when upgrading the firmware fro 5.14 to 6. I tried to re-flash with 5.14 and the following log keeps coming up. 

00:00:00 CFG: Use defaults
00:00:00 Project sonoff2 Sonoff (Topic sonoff2, Fallback DVES_A7C27E, GroupTopic sonoffs) Version 5.14.0-2_4_1
00:00:00 WIF: Connecting to AP1 <MY-SSID> in mode 11N as sonoffx-xxxx...

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

nalinda6963, 2018-07-13:
@AalianKhan Thanks. I will check the documentation and try this later.

AceoStar, 2020-06-11:
I'm currently seeing this on NodeMCU

AceoStar, 2020-06-11:
I'm still seeing this issue on NodeMCU.
`#include <ESP8266HTTPClient.h>
#include "ESP8266WiFi.h"

char myssid[] = "Spaces are great til they are not";
char mypass[] = "password";

const char* Host = "www.unwiredlabs.com";
String endpoint = "/v2/process.php";

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  Serial.println("Setup done");

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(myssid);
  WiFi.begin(myssid, mypass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(".");
}
`