esp8266/Arduino #1343: We need a Wifi (STA mode) function allowing to connect on a hidden SSID

HuguesD, 2016-01-01:
At the moment the Wifi.mode(STA) and Wifi.begin function do not allow to connect on a hidden SSID. It is possible to do it in the core Espressif SDK. Will this possibility be implemented in a future release ? when ?
thx

Testato, 2016-01-01:
For me it work.
Are you tried ?
Whitch error do you receive ?

HuguesD, 2016-01-02:
I receive the wifi.status : WL_NO_SSID_AVAIL = 1
and then it loops waiting for the WL_CONNECTED = 3 value.

HuguesD, 2016-01-02:
Also the Arduino Wifi library reference documentation indicates that it does the Wifi library only works for visible SSID.

Testato, 2016-01-02:
post your code,
i have in this moment two ESP connected in STA on a HIdden SSID

HuguesD, 2016-01-02:
Here is the code I used. It comes from the Sparkfun tutorial, that I modified a bit:

// Include the ESP8266 WiFi library. (Works a lot like the
// Arduino WiFi library.)
#include <ESP8266WiFi.h>
// Include the SparkFun Phant library.
#include <Phant.h>

//////////////////////
// WiFi Definitions //
//////////////////////

char WiFiSSID[] = "xxx";
const char WiFiPSK[] = "yyy";
int i=0;   

/////////////////////
// Pin Definitions //
/////////////////////
const int LED_PIN = 5; // Thing's onboard, green LED
const int ANALOG_PIN = A0; // The only analog pin on the Thing
const int DIGITAL_PIN = 12; // Digital pin to be read

////////////////
// Phant Keys //
////////////////
const char PhantHost[] = "data.sparkfun.com";
const char PublicKey[] = "wpvZ9pE1qbFJAjaGd3bn";
const char PrivateKey[] = "wzeB1z0xWNt1YJX27xdg";

/////////////////
// Post Timing //
/////////////////
const unsigned long postRate = 30000;
unsigned long lastPost = 0;

void waitForKey()
{
  // Wait for a carriage return from the serial port before proceeding.
  Serial.print("Waiting for CR...");
  while (Serial.available() == 0) {
    if (Serial.read() == '\r') break;    // Exit loop if we get a CR
  }
  while (Serial.available() != 0) {
    Serial.read();  // Flush any remaining input
  }
  Serial.println("Proceeding.");
}

// Put this everywhere you want to wait for a Return:
//  waitForKey();

void initHardware()
{
  Serial.begin(9600);
  pinMode(DIGITAL_PIN, INPUT_PULLUP);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, LOW);
  // Don't need to set ANALOG_PIN as input, 
  // that's all it can be.
  
}

void connectWiFi()
{
  byte ledStatus = LOW;
  uint8_t mac[WL_MAC_ADDR_LENGTH];
  //WiFi.macAddress(mac);
  //Serial.println(" ");
  // Serial.print("MAC: ");
   //Serial.print(mac[0],HEX);
  // Serial.print(":");
  // Serial.print(mac[1],HEX);
   //Serial.print(":");
  // Serial.print(mac[2],HEX);
   //Serial.print(":");
  // Serial.print(mac[3],HEX);
  // Serial.print(":");
  // Serial.print(mac[4],HEX);
   //Serial.print(":");
  // Serial.println(mac[5],HEX);

  // Set WiFi mode to station (as opposed to AP or AP_STA)
  
  Serial.println("Connexion au Wifi...");
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  for (int i = 0; i < sizeof(WiFiSSID); i++ )  // Convertit le SSID en majuscules
    {
      //WiFiSSID[i] = WiFiSSID[i] & 0b11011111;
      WiFiSSID[i]= toupper(WiFiSSID[i]);
    }
    //Serial.println("Cast en majuscules du SSID:");
   // Serial.println( &WiFiSSID[0] );

  //Scan for networks and verifies WiFiSSID is available
  // WiFi.scanNetworks will return the number of networks found
 
 // int n = WiFi.scanNetworks();
 // Serial.print("scan done; SSID found: ");
 // if (n == 0)
  //  Serial.println("no networks found");
 // else
 // {
  //  for (int i = 0; i < n; ++i)
  //  {
      // Print SSID and RSSI for each network found
   //   if (WiFi.SSID(i)== "BTS1CESTICI")
   //     Serial.println(WiFi.SSID(i));   
  //    }
   //     delay(2000);
  //  }
  
  
  // WiFI.begin([ssid], [passkey]) initiates a WiFI connection
  // to the stated [ssid], using the [passkey] as a WPA, WPA2,
  // or WEP passphrase.
  
 // Serial.print("Login: ");
 // Serial.println(WiFiSSID);
 // Serial.print("PWD: ");
 // Serial.println(WiFiPSK);
 // Serial.print("valeur de la variable WL_CONNECTED: ");
 // Serial.println(WL_CONNECTED);
 // waitForKey();
  // Use the WiFi.status() function to check if the ESP8266
  // is connected to a WiFi network.
    
  WiFi.begin(WiFiSSID, WiFiPSK);
  while (WiFi.status() != WL_CONNECTED)
  {
    // Blink the LED
    Serial.println(WiFi.status());
    digitalWrite(LED_PIN, ledStatus); // Write LED high/low
    ledStatus = (ledStatus == HIGH) ? LOW : HIGH;

    // Delays allow the ESP8266 to perform critical tasks
    // defined outside of the sketch. These tasks include
    // setting up, and maintaining, a WiFi connection.
    delay(100);
    // Potentially infinite loops are generally dangerous.
    // Add delays -- allowing the processor to perform other
    // tasks -- wherever possible.
  }
  Serial.println("connecté au SSID!");
}

int postToPhant()
{
  // LED turns on when we enter, it'll go off when we 
  // successfully post.
  digitalWrite(LED_PIN, HIGH);

  // Declare an object from the Phant library - phant
  Phant phant(PhantHost, PublicKey, PrivateKey);

  // Do a little work to get a unique-ish name. Append the
  // last two bytes of the MAC (HEX'd) to "Thing-":
  uint8_t mac[WL_MAC_ADDR_LENGTH];
  WiFi.macAddress(mac);
  String macID = String(mac[WL_MAC_ADDR_LENGTH - 2], HEX) +
                 String(mac[WL_MAC_ADDR_LENGTH - 1], HEX);
  macID.toUpperCase();
  String postedID = "Thing-" + macID;

  // Add the four field/value pairs defined by our stream:
  phant.add("id", postedID);
  phant.add("analog", analogRead(ANALOG_PIN));
  phant.add("digital", digitalRead(DIGITAL_PIN));
  phant.add("time", millis());

  // Now connect to data.sparkfun.com, and post our data:
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(PhantHost, httpPort)) 
  {
    // If we fail to connect, return 0.
    Serial.println("failed to connect");
    return 0;
  }
  // If we successfully connected, print our Phant post:
  client.print(phant.post());
  Serial.println("Connected to phant...");

  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line); // Trying to avoid using serial
  }

  // Before we exit, turn the LED off.
  digitalWrite(LED_PIN, LOW);

  return 1; // Return success
}

void setup() 
{
  
  initHardware();
  connectWiFi();
  digitalWrite(LED_PIN, HIGH);
}

void loop() 
{
  if (lastPost + postRate <= millis())
  {
    if (postToPhant())
      lastPost = millis();
    else
      delay(100);    
  }
}

Testato, 2016-01-02:
you forgot the foirst two include,

but i can confirm that there is no problem to connect to a hidden SSID. This esp8266 core do not know if the SSID is hidden or not, because you set the name and the password manually.
So simply by:
```
WiFi.begin(ssid, pass);

tablatronix, 2016-11-16:
I get this error when trying to connect to adhoc SSID also, not sure if related.

sbts, 2017-05-27:
@tablatronix adhoc networks are an entirely different kettle of nasty rotting fish.
There are many many devices that won't connect to adhoc networks, including some of the Mobile Phones and Tablets out there

devyte, 2017-10-19:
The capability is present. Closing.

closed 2017-10-19 by devyte