esp8266/Arduino #1349: how can i use the md5 library?

piux, 2016-01-02:
is it implemented?

i see theres a md5 library but no examples in how to use it?

sticilface, 2016-01-02:
Here are the functions
```
    void begin(void);
    void add(uint8_t * data, uint16_t len);
    void add(const char * data){ add((uint8_t*)data, strlen(data)); }
    void add(char * data){ add((const char*)data); }
    void add(String data){ add(data.c_str()); }
    void addHexString(const char * data);
    void addHexString(char * data){ addHexString((const char*)data); }
    void addHexString(String data){ addHexString(data.c_str()); }
    bool addStream(Stream & stream, int len);
    void calculate(void);
    void getBytes(uint8_t * output);
    void getChars(char * output);
```

example

```
MD5Builder md5;
md5.begin();
md5.add("blah blah blah blah blah blah");
md5.calculate();
Serial.println(md5.toString()); // can be getChars() to getBytes() too. (16 chars) .. see above 
```

quite easy.

piux, 2016-01-02:
and how can you change the key?

penfold42, 2016-01-02:
What key ?

It's just as hash that's barely better than sum(1) for crypto purposes.

piux, 2016-01-02:
well i  tought that md5 had a key to cypher and uncypher, im new to this concept sorry. i guess there is no way to uncypher incoming text.

closed 2016-01-02 by piux

francwalter, 2022-07-20:
Thank you, this (closed) issue should be somewhere in the help or so 😄