esp8266/Arduino #1351: consider enabling exception handling

grahamehorner, 2016-01-02:
please consider enabling exception handling, for better developer experience and allowing try { } catch(...) { } which would enhance overall stability and reduce code when developing libraries.  

'exception handling disabled, use -fexceptions to enable'

devyte, 2017-03-14:
I'm wondering if this will be implemented? I was rather surprised to find out exceptions are disabled.
Is there a specific reason for disabling?

smitthhyy, 2017-05-06:
Please look at enabling this. It should be part of the core. Thanks. PS: This is a vote to get it included ASAP! (Just in case that's not clear :) )

devyte, 2017-05-06:
I would like to point out that exception handling increases RAM usage. It
may not be appropriate for all use cases, so enabling it is not as simple
as flipping a switch.

besyildiz, 2018-02-20:
is not there any offer, how it is?

devyte, 2018-05-29:
Temporarily assigning to @d-a-v to put the issue on the map, per his exception playground.

artus9033, 2018-06-09:
Any progress on this?

ascillato, 2018-06-09:
Hi, I'm also very interested on this.

d-a-v, 2018-06-09:
There is some work about that in https://github.com/d-a-v/esp-quick-toolchain/

earlephilhower, 2019-01-16:
Exceptions now a menu option, closing.

closed 2019-01-16 by earlephilhower