esp8266/Arduino #1352: sntp.h patch to allow C++ compiler

dannybackx, 2016-01-02:
Hi,

This appears to be missing from sntp.h and also the other include files.

  Danny

acer: {541} diff -c sntp.h~ sntp.h
*** sntp.h~     2015-11-24 09:50:33.000000000 +0100
--- sntp.h      2016-01-02 15:45:41.976664979 +0100
***************
*** 7,12 ****
--- 7,17 ----
  #else
  #include "ip_addr.h"
  #endif
+ 
+ #if defined(__cplusplus)
+ extern "C" {
+ #endif
+ 
  /**
   * get the seconds since Jan 01, 1970, 00:00 (GMT + 8)
   */
***************
*** 66,68 ****
--- 71,77 ----
  #define sntp_servermode_dhcp(x)
  
  #endif
+ 
+ #if defined(__cplusplus)
+ }
+ #endif

igrr, 2016-02-29:
Since we are modifying some of the SDK headers anyway, it might make sense to add cpp guard blocks to all of them.

devyte, 2017-11-20:
@igrr exactly which SDK files should get the guards? Should I just add the guards to the files, or should there be some patch made to apply to newer versions in the future? I seem to remember you mentioning that you keep a set of patches somewhere...

dannybackx, 2017-11-20:
I know I need it for <sntp.h> but I believe most of the includes need it.

Some input :
acer: {43} fgrep extern *h
airkiss.h:extern "C" {
at_custom.h:extern uint8 at_customLinkMax;
mesh.h:extern "C" {
osapi.h:extern int os_printf_plus(const char * format, ...) __attribute__ ((format (printf, 1, 2)));
spi_flash.h:extern SpiFlashChip * flashchip; // in ram ROM-BIOS
user_interface.h:       REASON_EXT_SYS_RST      = 6             /* external system reset */

so airkiss.h and mesh.h are ok, the others probably not.

What does this protect against ? Well, without the guard, function definitions will be interpreted as C++ so they'll get mangled.
Inspection of the libraries for e.g. a couple of functions in user_interface.h shows they're all C style :
acer: {38} xtensa-lx106-elf-nm ../lib/*.a >x
acer: {40} fgrep system_deep_sl x
          U system_deep_sleep
00000bc8 T system_deep_sleep
00000a00 T system_deep_sleep_instant
000001d4 T system_deep_sleep_local_2
00000c2c T system_deep_sleep_set_option
acer: {41} fgrep system_restart x
          U system_restart
          U system_restart
          U system_restart
          U system_restart_local
000003d8 T system_restart
00000174 T system_restart_core
00000650 T system_restart_enhance
00000290 W system_restart_hook
000002d4 T system_restart_local
          U system_restart_local
acer: {42} fgrep system_os_task x
          U system_os_task
00003da4 T system_os_task

Similar for the SPI functions :

acer: {47} fgrep spi_flash_ x
          U spi_flash_erase_sector
          U spi_flash_get_id
          U spi_flash_issi_enable_QIO_mode
          U spi_flash_read
0000036c T spi_flash_erase_sector
0000023c T spi_flash_get_id
000000c0 T spi_flash_get_unique_id
000004c0 T spi_flash_issi_enable_QIO_mode
00000400 T spi_flash_read
0000029c T spi_flash_read_status
00000004 T spi_flash_set_read_func
000003a4 T spi_flash_write
000002e8 T spi_flash_write_status

So my bet is you need to add the guards to all files except the two that already have them.

     Danny

igrr, 2017-12-27:
@devyte c++ guards should be added to all files in tools/sdk/include.

About updates to newer versions, don't worry about that for now. Changes have been fairly small recently and i was able to generate diffs from the SDK and apply them here, with minimal manual intervention. I'll document the process some time later.

devyte, 2017-12-30:
Closed via #4044 .

closed 2017-12-30 by devyte