esp8266/Arduino #1354: problems with WiFi.setPhyMode

onkelfunny, 2016-01-02:
```
Sketch 1:
void setup() {
	WiFi.disconnect(true);
	delay(2000);
	WiFi.begin("ABCD12345678", "12345678");
}
```

```
Sketch 2:
void setup() {
	WiFi.disconnect(true);
	delay(2000);
	WiFi.setPhyMode(WIFI_PHY_MODE_11B);
	WiFi.begin("ABCD12345678", "12345678");
}
```

The esp runs great with Sketch 1. After Flashing Sketch 2 i can't connect to the Router.
Because Sketch 2 not working i tried Sketch 1 and now this is not working.

After a complete Flash erase, Sketch 1 works without problems.

Links2004, 2016-01-02:
may the SDK stores the WIFI_PHY_MODE_11B setting,
does your router support WiFi mode B?

onkelfunny, 2016-01-02:
the router don't support mode b.

i ave tried to set mode g and mode n but without luck.

I think the idk stores the information... the question ist how i can reset the mode without full flash reset?

Links2004, 2016-01-02:
the sdk docs not write over it, you can try set it to 0 or
```ESP.eraseConfig();``` this will delete all SDK config

devyte, 2017-10-19:
I can't reproduce this with latest git and my router. Closing.

closed 2017-10-19 by devyte