esp8266/Arduino #1355: WiFi.scanNetworks() dont work

gerrylenz, 2016-01-02:
Source
git clone https://github.com/esp8266/Arduino.git esp8266

function
WiFi.scanNetworks();

don't find any networks

reaper7, 2016-01-07:
similar problem...
seems that something is wrong, the same example ESP8266WiFi->WiFiScan
sdk 1.3 found 5 AP:
```
scan start
scan done
5 networks found
1: TP_LINK_ABC (-70) 
2: HP-Print-1e-LaserJet 100 (-83) 
3: NETIASPOT-FF (-89)*
4: NETIASPOT2-FF (-88) 
5: FON_NETIA_ZZZ (-89) 
```
the same example on latest github version (sdk 1.5):
```
scan start
scan done
4 networks found
1: HP-Print-1e-LaserJet 100 (-83) 
2: NETIASPOT-FF (-89)*
3: NETIASPOT2-FF (-88) 
4: FON_NETIA_ZZZ (-89) 
```
it's looking like the latest version sdk or wifi lib do not show first network or skips it

igrr, 2016-01-07:
Oh, right, I think this was mentioned somewhere. Previously there was a bug
in the SDK, the first node of linked list was always empty. Looks like they
fixed it. Need to update our code as well.

marvinroger, 2016-01-09:
Yes, I can also confirm this behavior on 2.1.0-rc1.

tzapu, 2016-01-21:
this still seems to be the issue with the latest github as well, is that correct?

tzapu, 2016-01-22:
is the fix for this just in the example?
i am not adding +1 to the index and some networks, if not all are missing from int n = WiFi.scanNetworks(); ...

Links2004, 2016-01-22:
@tzapu I had misunderstand the problem.
but i not finde the skipping of the first result in the source that noted.

tzapu, 2016-01-22:
@Links2004 no problem.

It seems WiFi.scanNetworks(); is not working properly, not finding some or all of networks in latests sdks

I am not sure how to go about replicating it... for instance, i have my own access point that does not appear in the scan, but if i enter ssid and pass manually, it works just fine

cheers

reaper7, 2016-01-29:
WiFi.scanNetworks() still do not working properly :(

WifiScan example show 5 networks - do not show(or find?) TP-LINK_XXXXXX network.

But this network are in range because ESP can connect to it (without any problems!) with other examples like WiFiWebServer etc!
As You see, windows machine find TP-LINK_XXXXXX network too, and this is network with strongest signal strength.

look at screenshot:
[image]

Links2004, 2016-01-29:
@igrr can you fix it? 
i not find the offset you noted here:
https://github.com/esp8266/Arduino/issues/1355#issuecomment-169633684
in our git.

reaper7, 2016-01-29:
it looks like first found network are "under index 0"

igrr, 2016-01-29:
Looks like the code has changed — I was referring to an older version which I wrote initially.
Will take a look.

reaper7, 2016-02-04:
now works ok :)
TNX!

tzapu, 2016-02-04:
i can also confirm it s working, thank you very much

pieman64, 2016-02-10:
Thanks for the scan fix which I think I have completed. It seems better now but I'm not 100% convinced I have it right. One router is 6" from my PC and reads 100%, the main router next door is 76% but when I keep scanning over and over again I see one or the other plus lots of 20% to 50% routers but never both of mine together. That suggests I might still be missing the first index in the scan or ESP's are a little crazy. The closest router is a bridge from the main router so I guess that could be the reason it only sees one but they do have different SSID's. Also with the 100% router switched off it takes 5 to 10 rescans to find the 76% router even though each scan is finding 6 or 7 20% to 50% routers. Is this normal behaviour from the ESP's or am I still missing the first index?

pieman64, 2016-02-10:
After around 70 or 80 scans I finally see both my routers. Strange little things these ESP's.

[image]

Links2004, 2016-02-10:
your scan look like there are many routers out there,
do you ever see more then 12 at once?
may there is a limit in the SDK how many they scan.

tzapu, 2016-02-10:
i am not 100% positive, but i think i have seen more than 12 ....
i hope i am not wrong

tzapu, 2016-02-10:
i was wrong, i can test
i am seeing 14 APs currently

pieman64, 2016-02-10:
12 is about the most I have seen but each scans seems to bring up different AP's. Or perhaps I should say not exactly the same 6 to 12 it found on the previous scan. Sometimes very different and sometimes missing both of my 100% and 75% AP's.

pieman64, 2016-02-10:
Just seen 13 AP's, none of which were my AP's.

tablatronix, 2016-02-19:
Could it be channel collisions or a specific mode, I remember seeing someone having an issue with B routers. Channel collisions ?

Mine fluctuates from 22- 27 aps

```C
/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);
  // Serial.setDebugOutput(true);
  
  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  delay(100);

  Serial.println("Setup done");
  Serial.println();
  Serial.println("MAC: " + WiFi.macAddress());
}

String encryptionTypeStr(uint8_t authmode) {
    switch(authmode) {
        case ENC_TYPE_NONE:
            return "NONE";
        case ENC_TYPE_WEP:
            return "WEP";
        case ENC_TYPE_TKIP:
            return "TKIP";
        case ENC_TYPE_CCMP:
            return "CCMP";
        case ENC_TYPE_AUTO:
            return "AUTO";
        default:
            return "?";
    }
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks(false,true);
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {

    // sort by RSSI
    int indices[n];
    for (int i = 0; i < n; i++) {
      indices[i] = i;
    }
    for (int i = 0; i < n; i++) {
      for (int j = i + 1; j < n; j++) {
        if (WiFi.RSSI(indices[j]) > WiFi.RSSI(indices[i])) {
          //int temp = indices[j];
          //indices[j] = indices[i];
          //indices[i] = temp;
          std::swap(indices[i], indices[j]);
        }
      }
    }

    Serial.print(n);
    Serial.println(" networks found");

    Serial.println("00: (RSSI)[BSSID][hidden] SSID [channel] [encryption]");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      // Serial.print(i + 1);
      Serial.printf("%02d", i + 1);
      Serial.print(":");

      Serial.print(" (");
      Serial.print(WiFi.RSSI(indices[i]));
      Serial.print(")");
      
      Serial.print(" [");
      Serial.print(WiFi.BSSIDstr(indices[i]));
      Serial.print("]");
      
      Serial.print(" [");
      Serial.print((String) WiFi.isHidden(indices[i]));
      Serial.print("]");

      Serial.print(" " + WiFi.SSID(indices[i]));
      // Serial.print((WiFi.encryptionType(indices[i]) == ENC_TYPE_NONE)?" ":"*");
      
      Serial.print(" [");
      Serial.printf("%02d",(int)WiFi.channel(indices[i]));
      Serial.print("]");

      Serial.print(" [");
      Serial.print((String) encryptionTypeStr(WiFi.encryptionType(indices[i])));
      Serial.print("]");

      Serial.println();
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(10000);
}
```

tablatronix, 2016-02-20:
I am also now seeing something similar after testing for a bit, for example my home router disappearing from scans now and then, maybe 2 out of 10 times

tablatronix, 2016-02-21:
Threw nodemcu on my chip and am not seeing this behavior. Ill keep looking into it.

tablatronix, 2016-02-27:
Well more testing and cannot reproduce now. Might be some channel interference, or fast scan switching modes, but my home ap is showing up in every scan now on 2 boards after thousands of scans. Will post details if it occurs again or I can isolate the operating mode

igrr, 2016-02-29:
This one is getting quite long, and the original bug has been fixed. @tablatronix please open a new issue if you are able to reproduce it.

closed 2016-02-29 by igrr

AnanthaChakravarthi, 2018-04-09:
hi everyone 
i am facing an issue with my esp8266.
it is showing 16 networks at a time and 9 at the other instant. when i check it with mobile there are more than 23 networks. can u suggest me what to do to get all the network ids.

aspiringguru, 2018-07-03:
AnanthaChakravarthi : I'm guessing different antenna & wifi chips will have different success rates at detecting wifi networks.  likely the esp8266 has poorer quality antenna and wifi chip than your mobile has.

I'm still curious if this is a hardware thing, or software delays needed to allow 'better' finding of wifi ssid's.

search2074, 2018-10-17:
Worked for me:
<details>
<summary>It's only helped</summary>
```
/*
    This sketch demonstrates how to scan WiFi networks.
    The API is almost the same as with the WiFi Shield library,
    the most obvious difference being the different file you need to include:
*/
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done, wifi routers: ");
  Serial.print(n);
  
  if (n == 0) {
    Serial.println("no networks found");
  } else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : "*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}
```
</details>