esp8266/Arduino #1356: can not config nodemcu as  access point

mkeyno, 2016-01-03:
hi guys , please help me out

I used the stable library of esp8266 Arduino and have no problem to set my nodemcu as  access point server , however after installing the staging version to get benefit of SPIFFS feature, can't config my ESP as  access point  and default or last saved SSID comes out ,I've tried couple of library  sample code , but no success , maybe it is not related to staging library at all, but yet have not figure out 
any help really appreciated

BR

Links2004, 2016-01-03:
in staging the wifi has a bug, 
stable will work and have SPIFFS too.

mkeyno, 2016-01-03:
thanks Link2004 , but does anyone report or discuss about this issue or should I return to previous library ? it is huge bug for the staging version should not ignore

mkeyno, 2016-01-03:
I find couple of same issue for staging version but wonder why still such issue has not been fixed  
https://github.com/esp8266/Arduino/issues/1054
https://github.com/esp8266/Arduino/issues/1103
https://github.com/esp8266/Arduino/issues/1123
https://github.com/esp8266/Arduino/issues/1141

Links2004, 2016-01-03:
the staging is older then the stable.
it fixed in stable and in the git version.

mkeyno, 2016-01-04:
tnx Links2004 it fixed, you guys are so awesome

closed 2016-01-04 by Links2004