esp8266/Arduino #1359: crash with version 2.0.0.

xnorbi, 2016-01-03:
Hello

My program going fine with no crash with version 1.6.5-947-g39819f0

When I using version 2.0.0. the program crash. My sketch is huge I cannot paste it, but the crash happans in this line:
```
Serial.println("1");
client->print(locsol[i].Name);
client->println("=");
Serial.println("2");
```

With this output into the serial console:
```
1

Panic C:\Users\Norbert\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266\core_esp8266_main.cpp:85 __yield

ctx: sys 
sp: 3fff18b0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fff1a60:  0000000d 3fff3d50 3fff27c0 40204df3  
3fff1a70:  0000000d 3fff27c0 3fff1ab0 40206640  
3fff1a80:  3ffe000a 3ffefd48 3fff27c0 402079c5  
...
...
3fff4ad0:  74697422 3a22656c 78652220 65732e74  
3fff4ae0:  726f736e 65742031 222e706d 6176222c  
3fff4af0:  4665756c 646c6569 22203a22 7
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```
**with version 2.0.0. the arduino says**:
Sketch uses 241,586 bytes (23%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 45,344 bytes (55%) of dynamic memory, leaving 36,576 bytes for local variables. Maximum is 81,920 bytes.

**with version 1.6.5-947-g39819f0 the arduino says**:
Sketch uses 322,532 bytes (30%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 60,156 bytes (73%) of dynamic memory, leaving 21,764 bytes for local variables. Maximum is 81,920 bytes.

I dont know is it important or not.

I include readelf and objdump files I dont know how read them. Also data captured from serial port (but there is many printed data there).
[objdump.txt](https://github.com/esp8266/Arduino/files/76613/objdump.txt)
[readelf.txt](https://github.com/esp8266/Arduino/files/76614/readelf.txt)
[capture.txt](https://github.com/esp8266/Arduino/files/76612/capture.txt)

igrr, 2016-02-29:
It seems like you are running these lines from a Ticker callback or from an interrupt handler.
You can set a flag from Ticker/interrupt handler and check the value of the flag in you main loop. If the flag is set, then clear it and perform the network tasks required.

devyte, 2017-10-19:
Closing as user error.

closed 2017-10-19 by devyte