esp8266/Arduino #1360: Arduino IDE Add-on : Serial Monitor for ESP8266

mytrain, 2016-01-03:
For the ones interested in I've recently developed an add-on for the ARDUINO IDE in order to be able to use the Serial Monitor with an ESP8266 generic module.

[image]
The add-on is available in the menu under 'Tools> ESP8266 Modify Serial Monitor' and starts a modified version of the Serial Monitor with 3 extra widgets:
- A 'clear' button to reset the content of the terminal
- A 'reset' check-box to set the status of the RTS signal (RESET)
- A 'programmation' check-box to set the status of the DTR signal

NOTE : Arduino Serial Monitor shall not be started otherwise modified one will not be available. Restart the IDE if you started the Arduino Serial Monitor before.

On start 'reset' is checked and 'programmation' is none that will reset the ESP8266 . Un-check 'reset' to get the board running.

[image]

If the community agrees and help me (I've never committed on GITHUB) I can publish the source code for the add-on.

Extract the following archive in the tools folder of your ARDUINO sketch folder.
http://mytrain.fr/cms//images/mytrain/private/ESP8266SM.zip

Duality4Y, 2016-01-03:
Why would you want this?
I am just curious.

mytrain, 2016-01-03:
Because DTR and RTS signals are not correctly managed by original serial monitor that keeps the ESP8266 board in reset mode once open.

penfold42, 2016-01-03:
I love it - it's a great idea and I'd love to see it included in the main code base

Duality4Y, 2016-01-04:
ah yea nice!
I reset my modules by hand so I don't see a particular need for it for me, but I can see where other might find it useful!

EUA, 2016-01-07:
Ooops!
````
Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: processing/app/debug/TextAreaFIFO
	at java.lang.Class.getDeclaredMethods0(Native Method)
	at java.lang.Class.privateGetDeclaredMethods(Class.java:2701)
	at java.lang.Class.getDeclaredMethod(Class.java:2128)
	at java.awt.Component.isCoalesceEventsOverriden(Component.java:6220)
	at java.awt.Component.access$500(Component.java:186)
	at java.awt.Component$3.run(Component.java:6174)
	at java.awt.Component$3.run(Component.java:6172)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.Component.checkCoalescing(Component.java:6171)
	at java.awt.Component.<init>(Component.java:6140)
	at java.awt.Container.<init>(Container.java:283)
	at java.awt.Window.<init>(Window.java:535)
	at java.awt.Frame.<init>(Frame.java:420)
	at javax.swing.JFrame.<init>(JFrame.java:233)
	at processing.app.AbstractMonitor.<init>(AbstractMonitor.java:26)
	at processing.app.AbstractTextMonitor.<init>(AbstractTextMonitor.java:40)
	at processing.app.SerialMonitor.<init>(SerialMonitor.java:37)
	at org.d2002.esp8266.arduino.serialmonitor.ESP8266SerialMonitor.<init>(ESP8266SerialMonitor.java:39)
	at org.d2002.esp8266.arduino.serialmonitor.ESP8266SM.createMonitor(ESP8266SM.java:120)
	at org.d2002.esp8266.arduino.serialmonitor.ESP8266SM.run(ESP8266SM.java:108)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.ClassNotFoundException: processing.app.debug.TextAreaFIFO
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 34 more
````

mytrain, 2016-01-07:
@EUA What is your Arduino IDE version ?
Mine is 1.6.5.
Anyway I'll investigate it this week-end

EUA, 2016-01-12:
@mytrain 1.6.7
But same on 1.6.6

itsjustvenky, 2016-01-12:
@mytrain thank you so much. This is a good one.

mytrain, 2016-01-12:
@EUA 
Are you using a window or a linux version.
If you are using a windows box please could you check if there is a pde.jar in your c:/program files/Arduino/lib folder.

Does anybody else got this issue ?

tontito, 2016-01-12:
i`m using windows and arduino 1.6.7 
c:\program files\Arduino\lib\pde.jar  available 400kb
same problem

mytrain, 2016-01-14:
@tontito
Could you please send me a listing of the entries contained in your pde.jar.
(rename it into pde.zip then use a ZIP archiver to get the list of entries)

tontito, 2016-01-15:
#  Arşiv C:\Arduino\lib\pde.jar
2015-12-17 16:23        Directory        Directory  META-INF
2015-12-17 16:23           103            90  META-INF\MANIFEST.MF
2015-12-17 16:23        Directory        Directory  cc
2015-12-17 16:23        Directory        Directory  cc\arduino
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\filters
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\libraries
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\libraries\filters
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\libraries\ui
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\packages
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\packages\filters
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\packages\ui
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\ui
2015-12-17 16:23        Directory        Directory  cc\arduino\contributions\ui\listeners
2015-12-17 16:23        Directory        Directory  cc\arduino\i18n
2015-12-17 16:23        Directory        Directory  cc\arduino\packages
2015-12-17 16:23        Directory        Directory  cc\arduino\packages\formatter
2015-12-17 16:23        Directory        Directory  cc\arduino\view
2015-12-17 16:23        Directory        Directory  cc\arduino\view\findreplace
2015-12-17 16:23        Directory        Directory  cc\arduino\view\preferences
2015-12-17 16:23        Directory        Directory  processing
2015-12-17 16:23        Directory        Directory  processing\app
2015-12-17 16:23        Directory        Directory  processing\app\forms
2015-12-17 16:23        Directory        Directory  processing\app\helpers
2015-12-17 16:23        Directory        Directory  processing\app\javax
2015-12-17 16:23        Directory        Directory  processing\app\javax\swing
2015-12-17 16:23        Directory        Directory  processing\app\javax\swing\filechooser
2015-12-17 16:23        Directory        Directory  processing\app\macosx
2015-12-17 16:23        Directory        Directory  processing\app\syntax
2015-12-17 16:23        Directory        Directory  processing\app\tools
2015-12-17 16:23          2952          1435  cc\arduino\ConsoleOutputStream.class
2015-12-17 16:23          1991          1033  cc\arduino\UpdatableBoardsLibsFakeURLsHandler.class
2015-12-17 16:23          5422          2440  cc\arduino\contributions\BuiltInCoreIsNewerCheck.class
2015-12-17 16:23          5310          2200  cc\arduino\contributions\ContributionsSelfCheck.class
2015-12-17 16:23           846           410  cc\arduino\contributions\filters\NoopPredicate.class
2015-12-17 16:23          1571           648  cc\arduino\contributions\libraries\LibraryByTypeComparator.class
2015-12-17 16:23          1342           604  cc\arduino\contributions\libraries\LibraryOfSameTypeComparator.class
2015-12-17 16:23          1466           740  cc\arduino\contributions\libraries\LibraryTypeComparator.class
2015-12-17 16:23          1257           582  cc\arduino\contributions\libraries\filters\CategoryPredicate.class
2015-12-17 16:23          1984           821  cc\arduino\contributions\libraries\filters\InstalledLibraryPredicate.class
2015-12-17 16:23          1105           511  cc\arduino\contributions\libraries\filters\OnlyUpstreamReleasePredicate.class
2015-12-17 16:23          3275          1261  cc\arduino\contributions\libraries\filters\UpdatableLibraryPredicate.class
2015-12-17 16:23          3294          1421  cc\arduino\contributions\libraries\ui\ContributedLibraryReleases.class
2015-12-17 16:23          2050           801  cc\arduino\contributions\libraries\ui\ContributedLibraryReleasesComparator.class
2015-12-17 16:23          1098           502  cc\arduino\contributions\libraries\ui\ContributedLibraryTableCell$1.class
2015-12-17 16:23         16219          7217  cc\arduino\contributions\libraries\ui\ContributedLibraryTableCell.class
2015-12-17 16:23          1195           524  cc\arduino\contributions\libraries\ui\DropdownInstalledLibraryItem.class
2015-12-17 16:23          1367           626  cc\arduino\contributions\libraries\ui\DropdownLibraryOfCategoryItem.class
2015-12-17 16:23          1347           623  cc\arduino\contributions\libraries\ui\DropdownLibraryOfTypeItem.class
2015-12-17 16:23          1201           527  cc\arduino\contributions\libraries\ui\DropdownUpdatableLibrariesItem.class
2015-12-17 16:23          7795          3178  cc\arduino\contributions\libraries\ui\LibrariesIndexTableModel.class
2015-12-17 16:23          1432           597  cc\arduino\contributions\libraries\ui\LibraryManagerUI$1.class
2015-12-17 16:23          2353           945  cc\arduino\contributions\libraries\ui\LibraryManagerUI$2.class
2015-12-17 16:23         12829          4745  cc\arduino\contributions\libraries\ui\LibraryManagerUI.class
2015-12-17 16:23          1255           585  cc\arduino\contributions\packages\filters\CategoryPredicate.class
2015-12-17 16:23          3396          1273  cc\arduino\contributions\packages\filters\UpdatablePlatformPredicate.class
2015-12-17 16:23          1099           504  cc\arduino\contributions\packages\ui\ContributedPlatformTableCell$1.class
[... cut ...]
2015-12-17 16:23           751           390  processing\app\tools\MenuScroller$MenuScrollTimer.class
2015-12-17 16:23          1306           583  processing\app\tools\MenuScroller$MouseScrollListener.class
2015-12-17 16:23          9488          3636  processing\app\tools\MenuScroller.class
2015-12-17 16:23           246           186  processing\app\tools\Tool.class
2015-12-17 16:23          4138          2271  processing\app\tools\ZipDeflater.class
#
# Sum                    Size         Packet  Files
#                       745684        361064  302

mytrain, 2016-01-22:
@tontito
Sorry for the delay. I finally understood the issue. 
Arduino code has been refactored somewhere between version 1.6.5 and 1.6.x.
I corrected my plugin hoping that it will work with both versions.
download url is remained unchanged
http://mytrain.fr/cms//images/mytrain/private/ESP8266SM.zip

Thanks for keep me updated

EUA, 2016-02-07:
I am using linux version.
Last version is working properly.
It's make my life easier.
Thank you very much.

asianremedy, 2016-02-28:
Hey,

Very nice plugin. Could you please post source code for this plugin? Actually If it's easier for you could you just remove check-box from reset? I need it for my application. Thanks in advance.

putyn, 2016-03-30:
@mytrain got this error in 1.6.8
```
Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.I18n._(Ljava/lang/String;)Ljava/lang/String;
	at org.d2002.esp8266.arduino.serialmonitor.ESP8266SM.run(ESP8266SM.java:102)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
got it working on an older version of the IDE i think is 1.6.5

OS windows 7 - 64bit

mytrain, 2016-03-31:
I'll try it on 1.6.7.

The 22nd of January I posted an updated version on same link than first one. Could you confirm you downloaded yours after the 22nd  (ESP8266SM.jar dated on 22nd of January with a size of 12 492 bytes).

mytrain, 2016-03-31:
I've got the same exception with Arduino 1.6.8.
Please upload the version 3 of the plugin 
[http://mytrain.fr/cms//images/mytrain/private/ESP8266SM.v3.zip] that works now in 1.6.8

putyn, 2016-03-31:
@mytrain i will give it a try when i get home and report back

EUA, 2016-06-25:
Looks working also with 1.6.9
This issue could be closed.

icanic, 2017-02-09:
Can someone explain me where to extract the zip file. 
I am using windows 10 with arduino IDE 1.8.1 and I have tryed putting it in some folders but theres nothing showing under the Tools tab.
I am using board NodeMCU 1.0 (ESP-12E Module). When I click on Show sketch folder I get this path "C:\Users\icani\OneDrive\Document\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ArduinoOTA\examples\BasicOTA"

kirchnet, 2017-03-15:
Same problem here as @icanic with the port error. OTA uploads work fine so clearly it's a setting related to the plugin.
I assume there is a combination of settings under the Tools menu, but which combination works? Presumably "Debug port" needs to be set to "Serial", but none of the settings under "Debug level" work.

devyte, 2017-10-19:
@mytrain could you please put it up in a github repo to have it under version control?

mytrain, 2017-10-19:
I'll do it tonight

----- Mail original -----

devyte, 2018-01-05:
@mytrain did you ever put it on github?

mytrain, 2018-01-05:
SOrry, I completly forget to do it.
Now it's done !!
Enjoy 2018

devyte, 2018-02-28:
The referenced upload seems to be [this one](https://github.com/mytrain/arduino-esp8266-serial-plugin).

devyte, 2018-02-28:
Pending: add a link to that repo from the docs.

bob5731, 2018-06-17:
how do install on a mac?

cheungbx, 2018-09-04:
I am Also looking for solutions for serial monitor not working when I boot it up in operate mode. Ie Chip Enable pulled high. It works when i get it to the program mode Aja flash mode (GOIO0 grounded), then once the  code upload is five and the chip go into operate mode by itself, the serial monitor will work and my message will be displayed. Will this help if I only connect the RX and TX 3.3V and ground from ESP 8266 generic module to the serial interface if my PC ?

cjdshaw, 2018-10-14:
> how do install on a mac?

Ctrl-click on Arduino.app and click "Show Package Contents"
Then put the ESP8266SM folder into Contents⁩/Java⁩/tools
so the full path should be
Arduino.app/Contents/Java/tools/ESP8266SM/tool/ESP8266SM.jar

PR #6744 (2019-11-10): Add some useful links
Fixes #1360
commit 43e0554b (2019-11-10): Add some useful links (#6744)

Fixes #1360
closed 2019-11-10 by devyte