esp8266/Arduino #1361: UART with 9n1

Digital-MoBa, 2016-01-03:
I wanna use the RS485 protocoll that communicate over UART. In this special case I need to send 9n1 (Databits = 9 and Stopbits = 1) at 62500 Baud. Is it possible to configurate the ESP to support this setting?

penfold42, 2016-01-03:
Can you hack the software serial library to suit your needs ?

Digital-MoBa, 2016-01-05:
I don't wanna change something insde the Arduino core. If other people wanna use this function, they need to do the same change. Is there a similar way for the ESP to add this in asm?
For the Arduino UNO I use that Assembler:
//--------------------------------------------------------------------------------------------
#include <avr/interrupt.h>

int XpressNetClass::USART_Receive(void)
{
	unsigned char status, resh, resl;
	// Wait for data to be received

	status = UCSRA;
	while (!(status & (1 << RXC))) { return -1; }//status = UCSRA;}

	// Get status and 9th bit, then data
	resh = UCSRB;
	resl = UDR;

	// If error, return -1 
	if (status & ((1 << FE) | (1 << DOR) | (1 << PE))) { return -1; }

	// Filter the 9th bit, then return 
	resh = (resh >> 1) & 0x01;
	return ((resh << 8) | resl);
}

//--------------------------------------------------------------------------------------------
void XpressNetClass::USART_Transmit(unsigned char data8) {
 // wait for empty transmit buffer

  while (!(UCSRA & (1<<UDRE))) {}
 // put the data into buffer, and send 
 UDR = data8;
}

penfold42, 2016-01-05:
Looking at the dock, the hardware uart will not do 9 bit.

You will have to bit bash it with softwareSerial.

Why don't you just create a SoftwareSerial9 library that's an almost direct copy of the existing library ?

Digital-MoBa, 2016-01-05:
Thanks for the fast answer. Yes to build up a SoftwareSerial9 library is a solution. 
I wanna realize the XpressNet protocoll with a translation to wifi and for this the master send always pakets over serial (each 120 microseconds - 9 bit request) and wait in this window for a client request. The transmit data size is up to 9 byte, in a transmission window of max. 500 microseconds.
I'm worry that the solution with the SoftwareSerial will block the wifi functions with it's interrupt requests. (While the SoftwareSerial read or send data the interrupt handle is off!)

tech4bot, 2016-02-06:
hi. did you found any solution to 9bit on esp8266, for 3 days I am searching for a solution but without any luck.

Links2004, 2016-02-06:
for sending (TX) only you can "use" SPI for example the SPI interface can send out any bit count you need.
the functions for this are not public in the class but its possible to write up to 512bit at once with SPI. 
https://github.com/esp8266/Arduino/blob/master/libraries/SPI/SPI.cpp#L219-L223

tech4bot, 2016-02-06:
can you, please, help me with an example?

thank you!

Links2004, 2016-02-07:
```cpp

void setup() {
    pinMode(MOSI, OUTPUT); ///< GPIO13
    digitalWrite(MOSI, 1); ///< Serial is Idle High
    SPI1C = 0;
    SPI.setFrequency(62500); 
    SPI1U = SPIUMOSI | SPIUDUPLEX | SPIUSSE;
    SPI1U1 = (10 << SPILMOSI) | (10 << SPILMISO); // set to 11 bit ( Start + Date (9) + End)
    SPI1C1 = 0;
}

void write9bitSerial(uint16_t data)  {
     data = ((~data) << 1) | 0x0001; // invert data  + add Start bit
     while(SPI1CMD & SPIBUSY) {}
     SPI1W0 = data;
     pinMode(MOSI, SPECIAL); ///< GPIO13
     SPI1CMD |= SPIBUSY;
     while(SPI1CMD & SPIBUSY) {}
     pinMode(MOSI, OUTPUT); ///< back to idle High
}
```
code not tested, but it shows the basics.

closed 2016-02-29 by igrr

i-make-robots, 2017-02-14:
Hello!  Sorry to resurrect this old thread.  I'm trying to read and write (1 start 8 data 1 stop) bits.  Would you please have an example of how to do this?  I don't know a better place to ask.

Digital-MoBa, 2017-02-14:
Hi,

I didn't solve this problem until now. What I know is that the hardware 
doesn't support this and you need to do it with software serial. So just 
add the additional start bit there and you have 9n1.