esp8266/Arduino #1363: ESP.deepSleep causes "hang" after wakeup?

dariopb, 2016-01-04:
ESP201, Arduino esp8266 (using platformio, but same happens if I compile using the arduino env), XDP pin connected to RST

void setup() {
  Serial.println("Setup");
  ESP.deepSleep(5 * 1000000);
}

Seems the chip enters deep sleep and then resets correctly, but:

ets Jan  8 2013,rst cause:5, boot mode:(3,7)
ets_main.c

and my setup code is not reached. 
How can I troubleshoot the issue?

Links2004, 2016-01-04:
gpio 16 is connected to ch_pd?

dariopb, 2016-01-04:
gpio 16 is connected to RST.

dariopb, 2016-01-04:
That was it! I went and looked and it seems the board i am using had **RST connected to CD_PD!** After removing this tie deep sleep and restart works as expected!  Thanks Links2004!

closed 2016-01-04 by dariopb