esp8266/Arduino #1366: Possible enhancement: allow SPIFFS.openDir to return false

andig, 2016-01-04:
Not a C expert, excuse me. While I can do 

	  File configFile = SPIFFS.open("/config.json", "w");
	  if (!configFile) {
	    Serial.println("Failed to open config file for writing");
	    return false;
	  }

I cannot do:

	  Dir dir = SPIFFS.openDir(path);
	  if (!dir) {
	    Serial.print("Failed to open directory ");
	    Serial.println(path);
	  }

due to:

	webserver.cpp:169: error: no match for 'operator!' (operand type is 'fs::Dir')
	sketch\webserver.cpp:169:7: note: candidate is:
	sketch\webserver.cpp:169:7: note: operator!(bool) <built-in>
	sketch\webserver.cpp:169:7: note:   no known conversion for argument 1 from 'fs::Dir' to 'bool'

Would it make sense to allow conversion for false similar to `open`?

igrr, 2016-01-04:
SPIFFS doesn't have directories, only file names. So doing
SPIFFS.openDir("/prefix") actually means "list all files which names begin
with "/prefix".
Handling an empty file list is easier than doing an explicit check, this is
why I think this method is not needed.

andig, 2016-01-04:
> SPIFFS doesn't have directories, only file names.

In that regard it makes sense, thank you!

closed 2016-01-04 by andig