esp8266/Arduino #1370: Servo read() does not return the expected value.

alvarolb, 2016-01-04:
Hi! I have been working with the servo library and I found an strange behaviour. For example, writing the position 120 will return in the read() method a 119. Writing a 121 will return in the read() 120... So I think it is not working properly as it is returning systematically the write value -1. The [Arduino documentation](https://www.arduino.cc/en/Reference/ServoRead) states that the read() method will return the previous value set in the write() function.

Here is a simple example:

```cpp
#include <Servo.h> 

Servo servo;

void setup(){
  Serial.begin(115200);
  servo.attach(D2);
}

void loop() {
  servo.write(120);
  Serial.println(servo.read());
  delay(5000);
  servo.write(121);
  Serial.println(servo.read());
  delay(5000);
}
```
The output result is like the following:

```
119
120
119
120
119
120
119
120
```

Meanwhile the expected value is 

```
120
121
120
121
120
121
120
121
```

penfold42, 2016-01-05:
A quick look over the code suggests it could be rounding issues because it uses uSec internally.

Writes and reads use map() to convert degrees to uS and back again.

alvarolb, 2016-01-09:
I will try to take a look on that...

Makuna, 2016-01-25:
Penfold42 is correct, it is a rounding error.  I will take a look at fixing that so it returns the same value you set it too.

penfold42, 2016-01-25:
It's a terrible calling convention - if < blah assume it's widgets else elephants.

Maybe a simple hack to cache the passed parameter to it can be returned ?

Ideally we'd have 2 set and 2 get methods but this might break existing code.

igrr, 2016-02-29:
Fix released in 2.0.0

closed 2016-02-29 by igrr