esp8266/Arduino #1372: Strange behavior on interrupts

vickeyhort, 2016-01-05:
Hi. I am using a two double channel relays with ESP-8266 12E. Please study attached picture for understanding my setup. I am using pubsubclient to publish and subscribe MQTT message to Mosquitto which is connected to Node-red and Domoticz. In short, four AC switches are being controlled from domoticz and also with 4 tactile push button switches with interrupt enabled. This setup works very stable (It's working since a week without any issue) if interrupts are disabled but when I enable interrupts, it behaves strange, if I want to turn relay one off, which is attached to GPIO13, relay four also turns off which is attached to GPIO5. If interrupts are disabled (I can enable and disable interrupts with MQTT messages), system works without any issue. May someone please help me with this problem?
[image]

Links2004, 2016-01-05:
you using the china relay boards?
if you switch loads with them they generate a EMC pulse, may it triggers your local switches.
i one setup they generate a reset of my esp, since then i only use solid state relais.
try to add some capacitors  to yous switches => 100nF.
a bigger cap for 3V3 >=47uF may also help 2x 100nF is a bit less for the ESP.

vickeyhort, 2016-01-05:
Yes I am using China boards, May you please suggest where should I add capacitors in switches? I have a solid power supply, and ESP8266 is running very fine, I just have that false interrupt trigger, How can I suppress the effect of EMC pulse? I have a lot of relays already, and I don't wanna waste these.

Links2004, 2016-01-05:
add between GPIOx and GND a ceramic capacitor (100nF or less).
adding a Pullup for D4 and D2 may also helps.
the EMC comes from 2 thinks: 
- the relay coil itself (always the same)
- the closing "switch" inside the relay (more load creates higher EMC pulse)

if you buy new stuff at on point, order solid state relays :)
for big load a good one is SSR-25DA or for load < 450W G3MB-202P both have zero crossing switching which reduce EMC too.

vickeyhort, 2016-01-05:
I also have SSR-10DA. May It run 2 horse power 240 v electric pump?

Links2004, 2016-01-05:
SSR-10DA can handle up to 2400W, how many Watt are "2 horse power" ?

vickeyhort, 2016-01-05:
2 hp = 1491 watts.
Thank you so much Links2004 for your kind support. but sorry I did not understand your previous comments about capacitors. As in picture above, I have four push button switches, which are connected at one end to gpio and to other end to GND, where should I put capacitor between them? Do you mean between gpio and GND of that particular switch prior to press button switch?

Links2004, 2016-01-05:
this is better understandable I think:
[image]

penfold42, 2016-01-05:
I've just built something similar and this is very timely.

How quickly does the ESP check the GPIOs for the boot mode on power up ?

The time constant here will mean gpio2 will read low for about a millisecond.

Links2004, 2016-01-05:
good point.
may add a 200nF to the Reset and it can not be a problem.
a lower C value may work the same, the needed size is depending on how strong the EMC is.
lowering R is a potion too, but lower then 1k is not recommended for a pullup.

igrr, 2016-02-29:
@vickeyhort has the issue been resolved for you?

vickeyhort, 2016-02-29:
Nope. Tried all mentioned methods but issue has not resolved yet. I am using two different ESP-8266 modules with two different power supplies, one for driving relays and other for button interrupts.

igrr, 2016-02-29:
Okay, please post the code which demonstrates the issue, so we can reproduce and fix it.

vickeyhort, 2016-03-01:
`                                                            /*
* Arduino code by Waqas Ahmed
* Follow the instruction given below
* Don't make changes in prohibted area

_________________Prohibted Area Start_______________________
                                                            */
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ESP8266httpUpdate.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>
#include "FS.h"

                                                            /*
_________________Prohibted Area Ended_______________________
It's user library attachment area!
It's user variable declaration area!
Area below can be editted:                                  */

const char* SKETCH_ID = "test";  //Non-overlaping name of sketch
const char* SKETCH_VERSION = "0.5"; // Sketch version
const char* MQTT_PUBLISH_TOPIC = "test-out"; // topic name to publish on mqtt
const char* MQTT_SUBSCRIBE_TOPIC = "test-in"; // topic to subscribe at mqtt for message reception

int NUMBER_OF_GPIOS = 2;  //Number of gpio's to be set as output
int NUMBER_OF_CUSTOM_VARIABLES = 6; //Number of custom variables defined
byte gpio[] = {13, 12}; //LED, Saver pin respectively
int on_state = 1;
int off_state = 0;

int GPIO_DEFAULT_STATE = off_state;
byte gpio_state[20];
byte custom_variable[20];

int led_pin = 13;
int saver_pin  = 12;
int door_pin = 14;
int pir_pin = 3;
int light_button_pin = 5;
int pump_button_pin = 4;

unsigned char* doorInterrupt = &custom_variable[0];
unsigned char* lightInterrupt = &custom_variable[1];
unsigned char* pumpInterrupt = &custom_variable[2];
unsigned char* electricity = &custom_variable[3];
unsigned char* sunset = &custom_variable[4];
unsigned char* activity = &custom_variable[5];

byte interrupt_pin[] = {5, 4}; //light, pump buttons repectively
int no_of_interrupts = 2;

boolean bounce = false; //Store bounce is active or not
long bounce_time = 0; //Time when bounce is activated

long activity_failsave = 0;
boolean door = false;
int door_check = 0;
int door_parked = 0;
long lastinterrupt = 0;

boolean lightsON = false;
long lightsON_time = 0;
boolean turnTheLights = false;
long last_pressed = 0;

boolean motion_sensed = false;
long motion_updated = 0;

float battery_correction = 0.10; //Add 0.10 to actuall reading
[... 688 lines cut ...]
  String cVar = "customVariable" + String(var);
  updateDb(cVar, val);
  send("custom_variable[" + String(var) + "] is set to " + String(val));
}

boolean send(String message)
{
  client.publish(MQTT_PUBLISH_TOPIC, message.c_str(), true);
}

boolean send(const char* topic, String message)
{
  client.publish(topic, message.c_str(), true);
}

String ipToString(IPAddress ip){
  String s="";
  for (int i=0; i<4; i++)
    s += i  ? "." + String(ip[i]) : String(ip[i]);
  return s;
}

                                                            /*
_________________Prohibted Area Ended_______________________
                                                            */`

sundeep110, 2017-01-19:
Hi HobbytronicsPK, 
I am facing exactly the same issue, i was interested in the software solution you proposed. However, by a glance at the code and your description above, I could see that we are assuming that only after 400ms the rising pin (5 in this case) is changing to high. 
However, for me this is not the case, the interrupt is raised immediately and pin changes its state immediately, I am interested if this software solution worked for you. Or please suggest if any other hardware or software solution worked for you.
Thanks in advance.

HobbytronicsPK, 2017-01-19:
I have tried number of solutions but I faced problems using interrupt. As
ESP8266 is very sensitive to EMI.

In my case, I used interrupts for toggle buttons and experienced issues. So
I just switched it towards reading pin's state in loop and perform
functions accordingly. Being ESP8266 much fast, It worked for me at least
better than those false triggers occurred while using interrupts.

or mute
the thread

vicnevicne, 2017-01-19:
@HobbytronicsPK What cabling (length and type of cable) do you have between the ESP and the switches ? If it is long and wires are close to each other, capacitive coupling between them causes very short peaks on the wires adjacent to the one of the pressed button, which are sufficient to trigger interrupts.
If you want to rule this out, can you place the buttons right next to the ESP ?

HobbytronicsPK, 2017-01-20:
I also tried your method by placing reed switch 8 feet apart. If both sides
of reed switch are close to each other then interrupt works fine by adding
above mentioned software solution.

However, same false trigger experiences if reed switch is kept under opened
state.

As far as type of cable is concerned, I also tried variety of cables and
all exhibited same results.

Even SSR relay also triggered false interrupt on turning load on or off
(Heavy load such as Water Pump).

All this EMI comes from AC source used as power supply to feed ESP8266. Try
feeding ESP8266 from battery being charged by solar charger etc.  may solve
this problem which I have not tried yet.

Just try reading input pin continously in loop (in place of interrupt
driven) You will get promising results.

duncan-a, 2017-01-20:
What value are the pullup/pulldown resistors on the inputs? Sounds like
something may be floating...

HobbytronicsPK, 2017-01-21:
10K Pullup resistors

vicnevicne, 2017-01-21:
I share Links2004 opinion that RC filtering is the most efficient way to solve the issue in hardware (for switch bounce, external EMI and crosstalk between lines altogether). My own experience (reed switches up to 10m away from my RPi-based alarm system) has shown that aggressive filtering was required to avoid spurious interrupts. I ended up with [this](https://www.screencast.com/t/SKcFP2mnud) and never got any issue since then...
Now software filtering is a valid option of course but unless that's the only task your ESP has to do, using a timer instead of a code called by interrupt might prove much more efficient (if you have bounces, the interrupt might be called tens of times in a row). To make sure it's not a resource hog, I suggest you temporarily add a counter incremented at the start of the interrupt. and print its value regularly.
For a detailed list of solutions, both hardware and software, see [this page](http://www.ganssle.com/debouncing-pt2.htm) (it's focused on debounce, but the symptoms are mostly the same).

ksundeep, 2017-01-21:
@HobbytronicsPK I am also using loop way instead of interrupts, thanks for the suggestion.

HobbytronicsPK, 2017-01-21:
No doubt, Interrupts are more efficient and filters do work but It always
causes trouble especially in case of using ESP8266 in environment with high
EMI. Good luck @ksundeep.

*W*aqas *A*hmed
00923007321041
*Malik Agriculture Farms*
*Mochiwali village, Wigha-mal road*
*Kahror Pakka, District Lodhran, Punjab*
*Islamic republic of Pakistan*

penfold42, 2017-01-21:
An optocoupler on the input might help too.

Long wires on relatively high impedence inputs would make a great antenna.

HobbytronicsPK, 2017-01-22:
As far as, I have investigated the issue, EMI was coming from power source
used to power up ESP8266.

Using filtering capacitors at power source ranging from 22uF to above 400uF
did not even solved my problem.

devyte, 2017-10-19:
Closing per #3655 .

closed 2017-10-19 by devyte