esp8266/Arduino #1374: 2.0.0 stable: Can't connect to AP

andig, 2016-01-05:
Trying my working sketch on a new ESP I've noticed that none of my devices are able to connect to ESP when in AP mode (no PSK). Connection simply fails.

Kindly asking for a way here to provide debug information for further diagnosis. Wireless itself is working as station mode if fine when I supply credentials.

Links2004, 2016-01-05:
can you try staging version?
https://github.com/esp8266/Arduino#staging-version-
the WiFi is reworked in this version, and there is an easy way for debugging to: https://github.com/esp8266/Arduino/pull/1350#issuecomment-168384542

andig, 2016-01-06:
No luck with staging- compile fails:

    Board generic (platform esp8266, package esp8266) is unknown
    Error compiling.

andig, 2016-01-06:
Next stopper: can't compile with Debug->all and Debug Serial->Serial:

	In file included from C:\Users\a25058\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp:26:0:

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp: In member function 'wl_status_t ESP8266WiFiMulti::run()':

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.h:37:66: error: cannot pass objects of non-trivially-copyable type 'class String' through '...'

	 #define DEBUG_WIFI_MULTI(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp:130:25: note: in expansion of macro 'DEBUG_WIFI_MULTI'

				 DEBUG_WIFI_MULTI("[WIFI] SSID: %s\n", WiFi.SSID());

andig, 2016-01-06:
However, flashing with DEBUG=none I can connect to the AP now- fine. Should we use this issue to further track rc1 problems?

closed 2016-01-06 by andig