esp8266/Arduino #1375: Strange SSL connection failure

cottsak, 2016-01-06:
# tl;dr

What does `ssl->need_bytes=4621 > 3939` mean? I'm told it's generated from https://github.com/igrr/axtls-8266/blob/514b6685c5a84232caeed72dc5720c652bbe9f73/ssl/tls1.c#L1286 (thanks @Links2004) but I'm not sure how to take this investigation further.

# Can you reproduce it?

To reproduce my issue, use this sketch with the *Debug Level* set to *HTTPClient*:

```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

void setup() {

  Serial.begin(115200);

  Serial.print("Connecting to wifi");
  WiFi.begin("youknow", "whattodo");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("\r\nWiFi connected.");
  Serial.println("access point:");
  Serial.println(WiFi.SSID());
  Serial.println("ip address:");
  Serial.println(WiFi.localIP());

}

void loop() {
    // wait for WiFi connection
    if(WiFi.status() == WL_CONNECTED) {

        HTTPClient http;

        //http.begin("https://api.twilio.com/", "79 E7 4F C0 02 71 C8 11 4A 30 7C 14 DA 09 AE 66 AB BB 50 1B");
        http.begin("https://api.github.com/", "CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C");

        int result = http.POST("somedata=yo");
        // error: ssl->need_bytes=4621 > 3939

        Serial.println("status code: " + String(result));

        if(result > 0) {
          Serial.println("body:");
          Serial.println(http.getString());
        } else{
          Serial.print("FAILED. error:"); Serial.println(http.errorToString(result).c_str());
          Serial.println("body:");
          Serial.println(http.getString());
        }

        http.end();
    }

    delay(10000);
}
```

You'll notice that using `https://api.github.com/` everything works ok. Sure the request returns a `404` but the point is that the certificate SHA1 fingerprint matches and a SSL connection is made.

```
[HTTP-Client][begin] url: https://api.github.com/
[HTTP-Client][begin] host: api.github.com port: 443 url: / https: 1 httpsFingerprint: CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C
[HTTP-Client] connect https...
please start sntp first !
[HTTP-Client] connected to api.github.com:443
[HTTP-Client] https certificate matches
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
```

However with Twilio I get this strange `ssl->need_bytes=4621 > 3939` in the debug output and the connection fails. I know the fingerprint is correct so I don't know what is different. On further investigation the only difference between the two certs I can see is the key length: Twilio is 4096 and github is 2048.

Can someone please help me out with this?

```
[HTTP-Client][begin] url: https://api.twilio.com/
[HTTP-Client][begin] host: api.twilio.com port: 443 url: / https: 1 httpsFingerprint: 79 E7 4F C0 02 71 C8 11 4A 30 7C 14 DA 09 AE 66 AB BB 50 1B
[HTTP-Client] connect https...
please start sntp first !
ssl->need_bytes=4621 > 3939

[HTTP-Client] failed connect to api.twilio.com:443
[HTTP-Client][returnError] error(-1): connection refused
status code: -1
FAILED. error:connection refused
body:
[HTTP-Client][returnError] error(-4): not connected

[HTTP-Client][end] tcp is closed
```

Bear in mind regarding the `please start sntp first !`, I get this failure wether I include `configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");` or not.

cottsak, 2016-01-07:
@igrr Do you happen to know if the issue I'm having here is the same as https://github.com/esp8266/Arduino/issues/43#issuecomment-148507531 ?

cottsak, 2016-01-07:
@igrr Is this the "fragment size" problem you mention https://github.com/esp8266/Arduino/issues/43#issuecomment-163535888 ?

igrr, 2016-01-07:
Yes I think the cause is the same.
While we can have full 16k TLS fragment buffer when connection is already established, we can not have it while key exchange is in progress. Reason for this is that key exchange generates a lot of temporary data, which can overflow heap if we also allocate 16k buffer.
So I'm using a smaller buffer initially and then re-allocating it when connection is established. This initial size is probably okay for 2048 bit keys, but seems to be insufficient for 4096 bit keys. It has to be tuned in axTLS build.

cottsak, 2016-01-07:
@igrr So for now we can't use SSL for certs with RSA 4096 bit keys? Is that the short answer?

igrr, 2016-01-07:
Sort of, i think the limit is on the total certificate size. Certificate
size depends on key length, but that's not the only factor. I.e. I can't
guarantee it will work for every cert with 2048 bit key.

cottsak, 2016-01-07:
Right. And that's what https://github.com/igrr/axtls-8266/blob/514b6685c5a84232caeed72dc5720c652bbe9f73/ssl/tls1.c#L1283 check is about isn't it? The max. message size?

igrr, 2016-01-07:
Yes it checks if the message will fit into the buffer size. The line where
initial buffer size is set is in ssl_new:
ssl->max_plain_length = 1460*4;
You may try recompiling axTLS with this value increased to resolve the
issue.

cottsak, 2016-01-07:
@igrr How do I do that? Is there instructions somewhere on how to recompile the source?

igrr, 2016-01-07:
There is a makefile. Clone the axtls repository, add xtensa toolchain bin
directory to your PATH, and then run 'make' on the command line. Xtensa
toolchain should be somewhere inside Arduino15 directory (in AppData on
Windows, ~/Library/ on OS X).

cottsak, 2016-01-07:
@igrr Ok, I think I've done that. I'm assuming I want to use the built `ssl/tls1.o` file? Where do I move it so it's included in my sketch?

igrr, 2016-01-07:
You need to use the built libaxtls.a and replace
Arduino15/packages/esp8266/something/tools/sdk/lib/libaxtls.a with it.
Sorry I'm not near my laptop atm so can't check the exact path. Then
compile the sketch again and the new library will be picked up.

qblacksheep, 2016-01-17:
Hi, could you tell me where you include this command " configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");"  to fix please start sntp first?

Links2004, 2016-01-17:
in the setup before the usage of ssl

cottsak, 2016-01-18:
@qblacksheep did you notice I mentioned that the `configTime()` didn't actually help with this problem?

qblacksheep, 2016-01-18:
I didn,t but in my case it helps.

cottsak, 2016-02-09:
@igrr I didn't manage to get the recompiling of axTLS to work. Any other ideas or thoughts as to whether we can work around these certificate lengths?

If not, does this then mean that esp simply won't be able to connect to some servers with SSL? Cos that would be a terrible shame right?

MarcusF2015, 2016-02-18:
I tried the example shown in the first posting and got this:

`Connecting to wifi.
WiFi connected.
access point:
MyWiFi
ip address:
192.168.179.91
[HTTP-Client][begin] url: https://api.github.com/
[HTTP-Client][begin] host: api.github.com port: 443 url: / https: 1 httpsFingerprint: CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C
[HTTP-Client] connect https...
[HTTP-Client] failed connect to api.github.com:443
[HTTP-Client][returnError] error(-1): connection refused
status code: -1
FAILED. error:connection refused
body:
[HTTP-Client][returnError] error(-4): not connected

[HTTP-Client][end] tcp is closed
pm open,type:2 0
`
Trying api.twilio.com or api.telegram.org: Same result. 
Arduino IDE 1.6.5 and 2.1.0-rc2

Links2004, 2016-02-18:
try to enable debug for SSL, may you get more info.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

MarcusF2015, 2016-02-18:
With "core+SSL":

`Connecting to wifi......
WiFi connected.
access point:
MyWiFi
ip address:
192.168.179.91
status code: -1
FAILED. error:connection refused
body:

pm open,type:2 0
`
With "core+WiFi" I got a compilation error:
`In file included from MyDisk\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp:26:0:
MyDisk\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp: In member function 'wl_status_t ESP8266WiFiMulti::run()':
MyDisk\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.h:37:66: error: cannot pass objects of non-trivially-copyable type 'class String' through '...'
 #define DEBUG_WIFI_MULTI(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )
                                                                  ^
MyDisk\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp:130:25: note: in expansion of macro 'DEBUG_WIFI_MULTI'
                         DEBUG_WIFI_MULTI("[WIFI] SSID: %s\n", WiFi.SSID());
                         ^
Fehler beim Kompilieren.`

Links2004, 2016-02-18:
you write above that you use 2.1.0-rc2 but the log shows 2.1.0-rc1.
check the dir ```MyDisk\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\```
if there still rc1 delete it and install rc2 new.

MarcusF2015, 2016-02-18:
Oh! Sorry!
Now I have deleted 2.1.0-rc1 and 2.0.0 in that directory and installed 2.1.0-rc2.
The result with "core+SSL" is the same like last posting. 
With "core+WiFi" it is:

`Connecting to wifiwifi evt: 0
wifi evt: 3
.
WiFi connected.
access point:
MyWiFi
ip address:
192.168.179.91
status code: -1
FAILED. error:connection refused
body:

pm open,type:2 0
`
It looks like SSL is not working, but there is not a headerfile missing?!?!

Links2004, 2016-02-18:
@MarcusF2015 try to set debug to "all"
@igrr is the debug function disabled in the current axtls binary?

MarcusF2015, 2016-02-18:
With "All":

`Connecting to wifiwifi evt: 0
wifi evt: 3
.
WiFi connected.
access point:
MyWiFi
ip address:
192.168.179.91
[HTTP-Client][begin] url: https://api.github.com/
[HTTP-Client][begin] host: api.github.com port: 443 url: / https: 1 httpsFingerprint: CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C
[HTTP-Client] connect https...
[HTTP-Client] failed connect to api.github.com:443
[HTTP-Client][returnError] error(-1): connection refused
status code: -1
FAILED. error:connection refused
body:
[HTTP-Client][returnError] error(-4): not connected

[HTTP-Client][end] tcp is closed
pm open,type:2 0
`

No message from SSL.

At other sketches using WIFIClientSecure I will get messages like
`ssl->need_bytes=... > ...`

Links2004, 2016-02-18:
we come at least to this line:
https://github.com/esp8266/Arduino/blob/6c8ca2d363c123500f964cb9fc0796e1effde17c/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L750-L751
but not sure whats goes wrong inside the connect.
try to add debug code inside
start here:

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L235-L239

MarcusF2015, 2016-02-18:
Modified WIFIClientSecure:
```
int WiFiClientSecure::connect(const char* name, uint16_t port) {
    DEBUGV("Name %s\r\n",name);
    DEBUGV("Port %s\r\n",port);
    if (!WiFiClient::connect(name, port))
        DEBUGV("Return 0\r\n");
        return 0;
    return 1;
}
```
This is the result:
```
[HTTP-Client][begin] url: https://api.github.com/
[HTTP-Client][be0000000, epc3=0x00000000, excvaddr=0x000001bb, depc=0x00000000
gin] host: api.github.com port: 443 url: / https: 1 httpsFingerprint: CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C
[HTTP-Client] connect https...
Name api.github.com

Port Fatal exception 28(LoadProhibitedCause):
epc1=0x4000228b, epc2=0x0
Exception (28):
epc1=0x4000228b epc2=0x00000000 epc3=0x00000000 excvaddr=0x000001bb depc=0x00000000

ctx: cont 
sp: 3fff1ee0 end: 3fff22b0 offset: 01a0

>>>stack>>>
3fff2080:  40001f46 00000004 feefeffe feefeffe  
3fff2090:  feefeffe feefeffe feefeffe feefeffe  
3fff20a0:  3fff2cb8 00000000 3ffea81c 40107428  
3fff20b0:  3fff2184 00000000 00000010 feefeffe  
3fff20c0:  feefeffe feefeffe feefeffe 40206af2  
3fff20d0:  00000000 3fff2140 3fff2140 3ffe912f  
3fff20e0:  40002514 3fffdd3c 3fff2480 3fff2d18  
3fff20f0:  00000001 0000000e 00000008 00000000  
3fff2100:  00000000 00000000 00000000 000001bb  
3fff2110:  3fff2d40 3fff2130 3fff2120 00000008 <
3fff2120:  3fff2140 3fff2140 00000004 40203bf4  
3fff2130:  000001bb 00000015 3fff2140 00000008 <
3fff2140:  3ffe9128 000001bb 00000000 00000001  
3fff2150:  3fff2c3e 3fff2b88 00000000 40107428  
3fff2160:  3fff2d18 3fff22fc 3fff2200 40204373  
3fff2170:  3fff2ce8 00000000 3ffea81c 3fff2c90  
3fff2180:  3fffdad0 0000000b 3fff2200 40204a0c  
3fff2190:  3fff2274 0000000b 00000010 40206b7a  
3fff21a0:  3fff2d10 4010743f 3fff21dc 40206af2  
3fff21b0:  3ffe99ac 0000000b 3fff2274 40206b4a  
3fff21c0:  3fffdad0 3ffe8c6c 3fff2274 3fff1288  
3fff21d0:  3fffdad0 00000000 3fff1280 40204ab8  
3fff21e0:  3ffe8ca8 00000000 3fff2200 40204ad6  
3fff21f0:  3fffdad0 00000000 3fff1280 402023af  
3fff2200:  3fff2d18 3fff2d18 3fff2480 0000000f  
3fff2210:  0000000e 000001bb 3f001388 3fff29f0  
3fff2220:  0000000f 00000001 3fff2201 3fff2d88  
3fff2230:  0000003f 0000003b 3fff2ca8 0000000f  
3fff2240:  00000000 3fff2cf0 0000001f 00000011  
3fff2250:  3fff2cd8 0000000f 00000000 00000000  
3fff2260:  00000000 00000000 ffffffff 3ffe9e00  
3fff2270:  00000000 3fff2c90 0000000f 0000000b  
3fff2280:  017aa8c0 3ffe9e80 5b7aa8c0 402054f0  
3fff2290:  3fffdad0 00000000 3fff1280 40204ef0  
3fff22a0:  feefeffe feefeffe 3fff1290 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Links2004, 2016-02-18:
```DEBUGV("Port %s\r\n",port); ```
port is uint16_t not char * use:
```DEBUGV("Port %d\r\n",port); ```

MarcusF2015, 2016-02-18:
Copy and paste is not clever. New result:
```
[HTTP-Client][begin] url: https://api.github.com/
[HTTP-Client][begin] host: api.github.com port: 443 url: / https: 1 httpsFingerprint: CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C
[HTTP-Client] connect https...
Name api.github.com Port 443

Return 0

[HTTP-Client] failed connect to api.github.com:443
[HTTP-Client][returnError] error(-1): connection refused
status code: -1
FAILED. error:connection refused
body:
[HTTP-Client][returnError] error(-4): not connected

[HTTP-Client][end] tcp is closed
pm open,type:2 0
```

Links2004, 2016-02-18:
follow the function calls until you finde the function that returns reason for the fail.
may try it with the git version for the core too.

MarcusF2015, 2016-02-18:
How to access to the members of `IPAddress remote_addr;` to print them with DEBUGV?

MarcusF2015, 2016-02-19:
Now it can resolve api.github.com. But there is another problem:

```
[HTTP-Client][begin] url: https://api.github.com/
[HTTP-Client][begin] host: api.github.com port: 443 url: / https: 1 httpsFingerprint: CF 05 98 89 CA FF 8E D8 5E 5C E0 C2 E4 F7 E6 C3 C7 50 DD 5C
[HTTP-Client] connect https...
Name api.github.com Port 443

IPAddress: 127.252.30.192

:ref 1

ssl/tls1.c:539 malloc 3944, left 32680

please start sntp first !
State:	sending Client Hello (1)
:wr

:sent 52

:rn 1436

:ww

:rd 5, 1436, 0

:rdi 1436, 5

:rd 74, 1436, 5

:rdi 1431, 74

State:	receiving Server Hello (2)
:rd 5, 1436, 79

2619331, left 31128

crypto/bigint.c:1072 realloc 1032, left 28392

:rdi 1357, 5

:rd 1352, 1436, 84

:rdi 1352, 1352

:c0 1352, 1436

:rn 1234

:rd 1225, 123to/bigint.c:1072 realloc 1032, left 24496

free 1073697584, left 24472

free -1339898992, left 24488

:rd 5, 1234, 1225

:rdi 9, 5

:rd 4, 1234, 1230

:rdi 4, 4

:c0 4, 1234

State:	receiving Server Hello Done (14)
4, 0

:rdi 1234, 1225

State:	receiving Certificate (11)
free -1157539545, left 31232

free 1912619331, left 31128

free 1912619331, left 31128

free 191free 1236675407, left 27864

free -1339898992, left 27424

free -1339898992, left 27320

crypcrypto/bigint.c:1072 realloc 1024, left 21488

free 174771, left 22048

free 1073701744, left 22072

free 1902044795, left 22600

free -1261887755, left 22888

free 1078111766, left 22912

free 1073696544, left 23464

free 1073702008, left 24520

free 1073704112, left 25072

State:	sending Client Key Exchange (16)
:wr

:sent 267

:ww

:wr

:sent 6

:ww

State:	sending Finished (16)
free 1073696544, left 25072

free 1073696544, left 25072

:wr

:sent 69

:ww

:rn 75

:rd 5, 75, 0

:rdi 75, 5

:rd 1, 75, 5

:rdi 70, 1

:rd 5, 75, 6

:rdi 69, 5

:rd 64, 75, 11

:rdi 64, 64

:c0 64, 75

State:	receiving Finished (left 26024

free 1073695112, left 26320

free -453830781, left 26344

free 1073695976, left 26616

free 1073698648, left 26640

free -1222899350, left 27160

free 1073694632, left 27184

free 66269000, left 28224

free -1148459954, left 28248

free 1953654083, left 28328

free -195631869, left 28568

free -2099839325, left 28600

free -1620975104, left 28632

free 2083371393, left 28928

free 1073697584, left 28960

free 1987320289, left 28976

free 1073699008, left 29288

free -1226821694, left 29312

free 1073699872, left 29584

free 1073698720, left 29872

free 1073695832, left 29896

free -1242839674, left 30936

free 73390063, left 30960

free 1073700168, left 31232

free 1073699328, left 31528

free 1073700440, left 31552

free 16707, left 31632

free 878994350, left 31752

free -1949403527, left 35704

ssl/tls1.c:1405 malloc 17408, left 18288

16)
free -951422434, left 25568

free -1157539545, left 25616

free 1912619331, left 25640

free 60251, left 25672

free 8421376, left 25672

free 1073698112, left 25936

free 865659732, left 25984

free -1833997568, left 26008

free 1073694736, [HTTP-Client] failed connect to api.github.com:443
[HTTP-Client][returnError] error(-1): connection refused
[HTTP-Client][returnError] tcp stop
:wr

:sent 53

:rn 1413

:ww

Alert: close notify
free -1339898992, left 36696

free 4751360, left 36920

:ur 1

:close

:del

status code: -1
FAILED. error:connection refused
body:
[HTTP-Client][returnError] error(-4): not connected

[HTTP-Client][end] tcp is closed
pm open,type:2 0
```

comino, 2016-02-22:
Any progress here? Have similar issues. 

Btw: How did you enable ssl debug output in the end? 
Your last comment was, it doesnt work, and it doesnt work work me too.

MarcusF2015, 2016-02-23:
Now it works. Not with 2.1.0-rc2. Only this the master branch and only with api.github.com. Not with api.telegram.org (2048 bytes certificate). The effect is the same as in the first posting from cottsak.
Choose debug Level "All". https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md
igrr said he could not guarantee it will work for every cert with 2048 bit key:
github (2048 bytes cert): working
telegram (2048 bytes cert): not working
twilio (4096 bytes cert): not working
There is too less memory for SSL.

cottsak, 2016-02-23:
@MarcusF2015 My conclusion too. Some 4096 and 2048 bit public keys push the memory limit. That's sad I think. esp could have been a really good player in keeping the IoT landscape secure. But unless it can be made to support any SSL key length, it might loose favour with those of us who take a "security first" approach.

@igrr Nothing can be done I guess hey?

igrr, 2016-02-23:
I think I got some twilio requests working at some point, so this may probably be solved by tweaking some parameters in TLS library (i.e. how much memory to reserve for connection negotiation stage). Not saying that *every* cert can be made to work though.

MarcusF2015, 2016-02-23:
- Is this right?: This repository contains Espressif SSL-Libraries an not igrr's version. 

- With 2.0.0 I can connect to telegram. The code is similar to [Giancarlo's Telegram Bot sketch](https://github.com/Gianbacchio/ESP8266-TelegramBot). Connecting every 2.5 sec I will get two or three WDT timeouts or exception (28) within 10 hours.
With 2.1.0.-rc2 I can't connect.

cottsak, 2016-02-23:
Another 2048 bit SSL key that works: maker.ifttt.com

MarcusF2015, 2016-02-23:
@cottsak Have you tried `api.telegram.org`?

cottsak, 2016-02-24:
@MarcusF2015 No I didn't. Do you have an example request?

Also, re the TLS params: I [started to do it above](https://github.com/esp8266/Arduino/issues/1375#issuecomment-169567673). Maybe that can help you.

MarcusF2015, 2016-02-24:
@cottsak Fine, I'll try it.
But:
`printf("ssl->need_bytes=%d > %d\r\n", ssl->need_bytes, ssl->max_plain_length+RT_EXTRA-BM_RECORD_OFFSET);`

`max_plain_length = 1460*4;`
`#define RT_EXTRA                    1024`
`#define BM_RECORD_OFFSET            5`

Calculate `max_plain_length+RT_EXTRA-BM_RECORD_OFFSET`:
The result is 6859 and you got 3939. Why?

cottsak, 2016-02-25:
@MarcusF2015 Sorry.. can't explain that mate. I think @igrr is your man.

closed 2016-02-27 by igrr

cottsak, 2016-02-28:
@igrr so what's the fix? Have you tweaked those tls values?

igrr, 2016-02-28:
Initial buffer size was increased. I tested with the sketch you posted above.

MarcusF2015, 2016-02-28:
@igrr Great work! Thank you!

DoctorWho8, 2016-02-28:
The fact that it worked is good. Remember some devices do not have the native intelligence to manage SSL. The fact that this device does is good to know. And igrr, you've done good here.

cottsak, 2016-02-29:
@igrr Thanks mate. Is this update only available on the staging feed for now?

igrr, 2016-02-29:
Only in stable (2.1.0).

noelgeorgi, 2016-03-15:
@igrr i am getting error compiling axtls-8266:
**xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Wall -Os -g -O2 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/include -Icrypto -Issl -I.  -c -o crypto/aes.o crypto/aes.c
crypto/aes.c:37:20: fatal error: string.h: No such file or directory
 #include <string.h>
                    ^
compilation terminated.
<builtin>: recipe for target 'crypto/aes.o' failed
make: *** [crypto/aes.o] Error 1**

igrr, 2016-03-15:
@noelgeorgi I think there is something wrong with the environment if `string.h` can not be found. At least in our latest CI everything is OK:
https://travis-ci.org/igrr/axtls-8266/builds/113109768

Please open an issue on axtls-8266 repository if you can not solve this. Please provide some details so we can help you, i.e. your OS, how you installed xtensa toolchain, can you build other programs with xtensa toolchain and so on.

igrr, 2016-03-15:
I'm locking this topic because the original issue has been fixed. Please open new issue if you have any concern.