esp8266/Arduino #1377: 2.1.0-rc1: no such board

andig, 2016-01-06:
No luck with staging on Windows, IDE 1.6.7 - compile fails:

    Board generic (platform esp8266, package esp8266) is unknown
    Error compiling.

Once this happens switching back to 2.0.0 doesn't work either. Only solution is to delete the rc1 folder.

Links2004, 2016-01-06:
try to delete the 2.0.0 and rc1 folder, and then install the rc1 new.

andig, 2016-01-06:
Fixed. Seems you can have either 2.0 or 2.1-rc1. Same for the boards manager btw. If both URLs are present, the stable versions (1st URL in my case) are suppressed- probably an IDE issue....

andig, 2016-01-06:
Next stopper: can't compile with Debug->all and Debug Serial->Serial:

	In file included from C:\Users\a25058\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp:26:0:

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp: In member function 'wl_status_t ESP8266WiFiMulti::run()':

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.h:37:66: error: cannot pass objects of non-trivially-copyable type 'class String' through '...'

	 #define DEBUG_WIFI_MULTI(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ )

	C:\Users\xx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc1\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp:130:25: note: in expansion of macro 'DEBUG_WIFI_MULTI'

				 DEBUG_WIFI_MULTI("[WIFI] SSID: %s\n", WiFi.SSID());

PyBerger, 2016-01-06:
Also had the DEBUG_WIFI_MULTI issue - commented and then was able to compile fine.

I have another issue, running in SoftAP mode, I seem to fail getting an IP address from the ESP.

my code 
```
IPAddress IP_AP(192,168,4,1);
IPAddress IP_Gateway(192,168,4,1);
IPAddress IP_Subnet(192,168,4,255);
Serial.setDebugOutput(true);
WiFi.mode(WIFI_AP);
WiFi.softAP(espSSID);
WiFi.softAPConfig(IP_AP, IP_Gateway, IP_Subnet);
```

What I'm getting with full debug activated :
```
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
add 1
aid 1
station: b4:07:f9:9d:6c:ee join, AID = 1
wifi evt: 5
station: b4:07:f9:9d:6c:ee join, AID = 1
wifi evt: 5
station: b4:07:f9:9d:6c:ee join, AID = 1
wifi evt: 5
.wifi evt: 7
wifi evt: 7
wifi evt: 7
```

When I just replace the SDK1.5.0 SDK by the 1.4.0 in the 2.1.0 release - It works again and I get the following traces 
```
wifi evt: 7
..chg_A3:-180
...wifi evt: 7
add 1
aid 1
station: b4:07:f9:9d:6c:ee join, AID = 1
wifi evt: 5
..WS:ac
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
:ur 2
```

Links2004, 2016-01-06:
DEBUG_WIFI_MULTI  issue see #1369
will check what the DHCP server has for a problem.

andig, 2016-01-16:
Thanks. Removing 2.0.0 did the trick.

closed 2016-01-16 by andig