esp8266/Arduino #1378: esp8266.github.io contains outdated OTA documentation

julianh2o, 2016-01-06:
As of posting this, this URL still contains information that suggests Arduino-PR-4107-BUILD-421 as a requirement for use of ArduinoOTA.
http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#arduinoota-configuration

This page, in contrast, has been updated since the pull was merged and accepted into the release for 1.6.7
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md

I've just successfully used the BasicOTA example using the second set of instructions with an unmodified version of Arduino 1.6.7 (mac), and the 2.0.0 version of the ESP8266 Arduino boards.

I'd also vote that we promote the "ArduinoOTA" method to being the "preferred" method sooner rather than later to encourage new developers to test it and report any troubles they run into.

igrr, 2016-01-07:
These docs are snapshots of markdown docs present in git at the time of
particular release. There is no way to selectively update parts of these
docs; rather, we can just publish new release. The change you have
mentioned is present in 2.1.0-rc1 and the respective set of docs.

marvinroger, 2016-01-07:
By the way, the HTTP OTA now has since 2.1 a parameter not to reboot automatically the ESP. This is not documented yet.

closed 2016-02-29 by igrr