esp8266/Arduino #1379: Issue with new black board ESP-01E 1MB flash size

mkeyno, 2016-01-06:
hi 
I used the ESP-01 pin 0 and pin 2 for digital reading or writing , I don't know where exactly I read but I pull up those pin with different resistor such as 10k, 40k, 82k and even 100k . for difference project different resistor has work and I've not yet figure out what is relation between booting sketch and this variety of resistors , however I fix it somehow when I used the old blue ESP-01 board but when I change to new black board the problem getting worst and some time never boot or when it booted repeatedly dumping memory come out and never stop , I don't know why there is no problem to upload the sketch but no reliable booting , it is related to the pull up resistor , or related to flash size selection(SPI_SIZE_MAP) in Arduino IDE?
any help really appreciated guys

my platform new version of  Arduino stable version

penfold42, 2016-01-06:
What are the projects ?

I had an issue driving an non transistor from the GPIOs. 
I had a 10k pull-up on the GPIOS but the transistor and its base resistor were stronger than the pull-up resulted in about 1v on the gpio which is not even close to the 75% threshold to read a high on startup.

mkeyno, 2016-01-06:
hi , thanks for reply 
for example I read the PIR sensor 5 volt level directly by GPIO2 connected with 100k resistor , other smaller resistor make situation like a short cut , however my problem is with new black ESP-01E board which is make very unstable and I don't know whether this situation related to GPIO pins or not

drmpf, 2016-01-06:
Try measuring the voltage at GPIO2 and GPIO0 with a multi-meter.  I had a similar problem with an external opti-isolator that should have been off, but there was enough leakage to make GPIO0, 1.5V

mkeyno, 2016-01-07:
it seems the ESP pins so sensitive to the noise but I was wonder why the new black version is so unreliable against analog read of external voltage , my module continuously get reset and dumping memory

igrr, 2016-02-29:
Pull-up resistor on GPIO2 should be 10k to 3.3V VCC. Keep in mind that ESP8266 pins are not 5V tolerant, so connecting them to 5V is only possible via a divider or a level shifter.

closed 2016-04-10 by mkeyno