esp8266/Arduino #1380: exception after post JSON Data

onkelfunny, 2016-01-06:
i receive this exception after post JSON Data to a ESP module. (i post the JSON 4 or 5 times)
```
E:M 408
E:M 1584
E:M 768
Fatal exception 28(LoadProhibitedCause):
epc1=0x4024395f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000100, depc=0x00000000

Exception (28):
epc1=0x4024395f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000100 depc=0x00000000

ctx: sys 

sp: 3ffffd20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffec0:  40243309 3ffffee0 00000004 40105850  
3ffffed0:  00000000 00000007 00000000 fffffffe  
3ffffee0:  3ffea390 00000300 00000000 00000000  
3ffffef0:  40206ea9 00000000 00000000 f7ffffff
```
```
40243954 <phy_check_data_table>:
40243954:    630b          addi.n    a6, a3, -1
40243956:    169c          beqz.n    a6, 4024396b <phy_check_data_table+0x17>
40243958:    070c          movi.n    a7, 0
4024395a:    028d          mov.n    a8, a2
4024395c:    a09620            addx4    a9, a6, a2
4024395f:    0858          l32i.n    a5, a8, 0
40243961:    884b          addi.n    a8, a8, 4
40243963:    757a          add.n    a7, a5, a7
40243965:    f69897            bne    a8, a9, 4024395f <phy_check_data_table+0xb>
40243968:    000046            j    4024396d <phy_check_data_table+0x19>
4024396b:    070c          movi.n    a7, 0
4024396d:    a02320            addx4    a2, a3, a2
40243970:    fed222            addmi    a2, a2, 0xfffffe00
40243973:    f37c          movi.n    a3, -1
40243975:    303730            xor    a3, a7, a3
40243978:    b48c          beqz.n    a4, 40243987 <phy_check_data_table+0x33>
4024397a:    7f2222            l32i    a2, a2, 0x1fc
4024397d:    140c          movi.n    a4, 1
4024397f:    c02230            sub    a2, a2, a3
40243982:    932420            movnez    a2, a4, a2
40243985:    f00d          ret.n
40243987:    7f6232            s32i    a3, a2, 0x1fc
4024398a:    020c          movi.n    a2, 0
4024398c:    f00d          ret.n
```

onkelfunny, 2016-01-10:
no errors with the new idk version

closed 2016-01-10 by onkelfunny

pjsg, 2016-04-25:
@onkelfunny I also see this crash -- which version of the SDK is this fixed in?

onkelfunny, 2016-04-25:
@pjsg hello. I use the git Version. I make a stress test this evening, maybe there is a new error.
You have also this error in the phy_check_data_table ?

pjsg, 2016-04-25:
I'm using the nodemcu firmware (which uses the underlying espressif code) and I get a crash in phy_check_data_table at the same offset. I'm using the Espressif SDK 1.5.1. 

If this crash has gone in 1.5.3 then that would be great news!